modmypi/PiModules #1: Using for first time

ronisegoly, 2017-01-05:
I just received UPS PIco HV3.0 A Stack 450
I connected to my Pi and started OK after following installation instructions

I se Blue led blinking slowly and Yellow Led for Battery
After few minutes system will suddenly shutdown (Blue becomes steady)   
Is it because battery is not charged?
Any interface I can see battery status? 
Any user manual for the buttons and the other leds?

PiModules, 2017-01-05:
Hi,

Please find my answer in the *text*

>
> I just received UPS PIco HV3.0 A Stack 450
*OK. Happy about that*
>
> I connected to my Pi and started OK after following installation 
> instructions
>
> I se Blue led blinking slowly and Yellow Led for Battery
*If it is slowly blinking means that system is running on battery. The 
most possible reason for that is that your PSU is bad, or your current 
load is too big for your PSU
If the battery LED is lit, then battery is low
*
>
> After few minutes system will suddenly shutdown (Blue becomes steady)
*You need better PSU*
>
> Is it because battery is not charged?
*YES*
>
> Any interface I can see battery status?
*Read the manual, and see the battery level register*
>
> Any user manual for the buttons and the other leds?
*Use the preliminary before we will release the official one

http://www.forum.pimodules.com/viewtopic.php?f=25&t=2275

BR ioannis
*

-- 
Ioannis Mourtsiadis
Electronic Engineer M.Sc

Pi Modules
Nikaias 49, Nea Smyrni
17122 Athens, Greece

Tel:    +30 211 0130067
Fax:    +30 211 8500132
Mobile: +30 697 7089430
SKYPE:  i_mourtsiadis

www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

ronisegoly, 2017-01-06:
Thanks
We'll go topic by topic
1. The battery is charged by the board or I need to charge separately? RPI was on all night, led is still on
2. I am not sure what's the meaning of   - "or your current load is too big for your PSU"

PiModules, 2017-01-06:
Hi,

In the *text*

>
> Thanks
> We'll go topic by topic
>
>  1. The battery is charged by the board or I need to charge
>     separately? RPI was on all night, led is still on
*Battery is charged  always by the board automatically however the 
following conditions must be:*
*1. System is cable powered**
**2. The system is running (UPS LED is blinking fast - NOT slow)**
**3. Powering voltages are within USB powering specifications (4.75 - 
5.25 VDC) *
>
>  1. I am not sure what's the meaning of - "or your current load is too
>     big for your PSU"
  * *If your PSU is bad, then the powering voltage is lower than
    specifications, so Battery Back up is activated, so system is not
    charged**(to avoid current loop - system would charge himself when
    powered by battery that is charged from himself)*
  * *If you powering cable is not good, any request for current by the
    system will cause voltage drop and automatic switch to battery
    supply (please avoid to use PSU will separated USB cable, as also
    PSU with lower current than 2.5A)
    *
  * *If you PSU is not able to supply enough current to the system, this
    will cause voltages drop, so this will cause also automatic
    switching to battery back up*

*
**THE CORE REQUIREMENTS is to have GOOD PSU and GOOD POWERING CABLE ABLE 
TO SUPLY THE SYSTEM with voltage and current specification on the micro 
USB powering entry like specified by the Raspberry Pi(R)  manufacturer 
-  raspberrypi.org**
**
**As we have already 2 running technical forums I would suggest you to 
move this discussion to one of them, to avoid opening another third one.**
**The first one is serviced by www.pimodules.com (me)  and the second 
one is serviced by www.modmypi.com**(Jake)

**Please choose what is closer to you and move this discussion to it.**

**My Warmest Regards**
**Ioannis*

ronisegoly, 2017-01-06:
I found the forums only after posting here, will move over

jjtronics, 2017-03-11:
Hello,
I have a problem to update my firmware ...
I have UPS PIco HV3.0A with : 
UPS PIco Firmware.....: 24
  UPS PIco Bootloader...: 53
  UPS PIco PCB Version..: 41

But if i would like to upgrade to : UPS_PIco_0x5C_25_02_2016.hex i have this error : 
pi@raspberrypi:~ $ sudo python 9600_picofuHV3.0.py -v -f UPS_PIco_0x5C_25_02_2016.hex
Validating firmware: OK
Checking communication with bootloader: OK
Uploading firmware: 0% **************************************************************************************** 4.0% ***************************************************************************************** 9.0% ***************************************************************************************** 14.0% ***************************************************************************************** 19.0% ***************************************************************************************** 24.0% ***************************************************************************************** 29.0% ***************************************************************************************** 34.0% ***************************************************************************************** 39.0% ***************************************************************************************** 44.0% ***************************************************************************************** 49.0% ***************************************************************************************** 54.0% ***************************************************************************************** 59.0% ***************************************************************************************** 64.0% ***************************************************************************************** 69.0% ***************************************************************************************** 74.0% ***************************************************************************************** 79.0% ***************************************************************************************** 84.0% ***************************************************************************************** 89.0% ***************************************************************************************** 94.0% ***************************************************************************************** 99.0% **** Done uploading...
Invoking factory reset of PIco...
send factory reset command ...
Traceback (most recent call last):
  File "9600_picofuHV3.0.py", line 545, in <module>
    FWUpdate()
  File "9600_picofuHV3.0.py", line 189, in __init__
    self.factory_reset()
  File "9600_picofuHV3.0.py", line 533, in factory_reset
    i2c.write_byte_data(0x6b, 0x00, 0xdd)
IOError: [Errno 5] Input/output error

and it's impossible to reupload the firmware, the i2c is not writable ...

I use the bottom to the UPSpico card to manualy upload the previously firmware : 
pi@raspberrypi:~ $ sudo python 9600_picofuHV3.0.py -v -f /home/pi/servicepack_HV3.0_docs_0x24/0x24_UPS_PIco_HV3.0_main.hex
Validating firmware: OK
Checking communication with bootloader: OK
Uploading firmware: 0% *********************************************************************************** 4.0% ************************************************************************************ 9.0% ************************************************************************************ 14.0% ************************************************************************************ 19.0% ************************************************************************************ 23.0% ************************************************************************************ 28.0% ************************************************************************************ 33.0% ************************************************************************************ 38.0% ************************************************************************************ 42.0% ************************************************************************************ 47.0% ************************************************************************************ 52.0% ************************************************************************************ 57.0% ************************************************************************************ 62.0% ************************************************************************************ 66.0% ************************************************************************************ 71.0% ************************************************************************************ 76.0% ************************************************************************************ 81.0% ************************************************************************************ 85.0% ************************************************************************************ 90.0% ************************************************************************************ 95.0% ****************************************************************************** Done uploading...
Invoking factory reset of PIco...
send factory reset command ...
ALL Done :) Ready to go...

And it's OK ....

I test with another RPI3 but i have the same problem with the new install of Raspbian ....

Have you any idea of this problem ?

Thank you in advance,

PiModules, 2017-03-11:
Hi,

You are uploading a wrong firmware. The firmware you are uploading is 
for the old UPS PIco HV1.1 or HV1.0 and _*NOT *_for the UPS PIco HV3.0
So, it is just not working after upload, because it is different processor
As you have destroyed the existing firmware, luckily not the bootloader, 
so after hand invoking of it, uploading of proper one, should be working 
properly also for command mode.
You can see that you are uploading different one one each command line 
you provided. The second one is proper.
You need to upload the proper firmware, from here

http://www.forum.pimodules.com/viewtopic.php?f=25&t=2691

BR Io

jjtronics, 2017-03-11:
Hi !

Thank you for you return,
I would like to use my UPS Pico HV3.0 with the Jeedom Plugin : https://jeedom.com/doc_old/documentation/plugins/upspico/fr_FR/upspico
But the minimal requirement is firmware 25 .
I imagine he is talking about old cards.

Thank you for you help !