modmypi/PiModules #2: UPS Pico constantly switching from battery to usb power

ericcastro, 2017-01-15:
I started using this and found what I think isn't normal behavior:

The piezo beeps continuosly every 6 or 7 seconds, doing a different tone each time. I believe I discovered this is because the UPS Pico is constantly switching from USB power to battery. When I run:

```sudo i2cget -y 1 0x69 0x00 b```

I get 0x02 most of the time even though it's powered via USB, and eventually I get 0x01 but I immediately get back to 0x02

Then if I disconnect the USB power, the raspberry shuts down.

Am I doing something wrong, or is the UPS Pico defective?

I uploaded a video to demonstrate this:
https://www.youtube.com/watch?v=yCWGeNUxF88

PiModules, 2017-01-15:
Hi,

It is normal if you are not using a proper PSU or cable supplying it 
(micro USB).

Please use much better PSU (recommend 3A@5.1V) with powering cable on 
one body with the PSU (avoid double USB connectors)

Please use one of the following forums for your technical support fro 
the UPS PIco HV3.0

http://www.forum.pimodules.com/viewforum.php?f=20

or

http://forum.modmypi.com/technical-support/

My Warmest Regards

Ioannis (from PiModules)

-- 
Ioannis Mourtsiadis
Electronic Engineer M.Sc

Pi Modules
Nikaias 49, Nea Smyrni
17122 Athens, Greece

Tel:    +30 211 0130067
Fax:    +30 211 8500132
Mobile: +30 697 7089430
SKYPE:  i_mourtsiadis

www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

ericcastro, 2017-01-16:
I was using an iPad (10W usb output) which I thought was enough (2.1A@5V). I will try using more amps later today.

Finally, does the UPS Pico support the 40pin header 5V pin as power input instead of the micro USB ?

I ask this because this is for a robot whose supply comes from a 12V source which connects to another HAT that will then send 5V through the 40 pin rpi header.

I need to know if I can continue to use that pin as power input or the only way UPS Pico works is if power is supplied via micro USB

Thanks

PiModules, 2017-01-16:
Hi,

Yes,  it is accepting supply via GPIOs, but you need to have ZVD (zero 
Voltage Drop) circuit on your power supply side (it is already on the 
RPi when powered via micro USB) to avoid back charging the PSU when is 
battery powered. If you use small current <=1A, can be just a 
schottkydiode , but remember that voltage on the 5V GPIO must be withing 
USB specs (4.75=5.25 V) after diode or ZVD (just on GPIO pins)

Please measure the voltages on the GPIO 5V pins.

BR Ioannis

-- 
Ioannis Mourtsiadis
Electronic Engineer M.Sc

Pi Modules
Nikaias 49, Nea Smyrni
17122 Athens, Greece

Tel:    +30 211 0130067
Fax:    +30 211 8500132
Mobile: +30 697 7089430
SKYPE:  i_mourtsiadis

www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

ericcastro, 2017-01-19:
I keep getting the constant beeping even though I moved to a 3 amps power supply.

```sudo i2cget -y 1 0x69 0x00 b```

keeps switching from 0x01 to 0x02 but returning 0x02 most of the time...

is there anything that can be done, or do I have a defective UPS Pico ?

PiModules, 2017-01-19:
Your PIco is not defective.

Do you have PIco Stack or Plus?

Please send me pictures of your PSU

also run the following command and send me results

sudo i2cget -y 1 0x69 0x08 w && sudo i2cget -y 1 0x69 0x0a w && sudo 
i2cget -y 1 0x69 0x24 &&sudo i2cget -y 1 0x69 0x25 && sudo i2cget -y 1 
0x69 0x26

BR Ioannis @ PiModules

PiModules, 2017-01-19:
I check your video, until now I replying you by email. The PIco is not keeping the system running because the battery is completely empty. You can see the LED BAT lighting. Can  you please also run the system on a clear RPI + PIco, nothing else, and report it to me.
BR Ioannis@PiModules

ericcastro, 2017-01-21:
it's a UPS PIco HV3.0A Stack with 450mAh Battery

there is no other hardware attached other than a USB dongle for the wireless keyboard.

the power supply I am using is [this one](https://www.amazon.fr/gp/product/B01566WOAG/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1) from amazon.

the output of the command you sent is:
```
pi@raspberrypi:~ $ sudo i2cget -y 1 0x69 0x08 w && sudo i2cget -y 1 0x69 0x0a w && sudo i2cget -y 1 0x69 0x24 && sudo i2cget -y 1 0x69 0x25 && sudo i2cget -y 1 0x69 0x26
0x0349
0x0484
0x41
0x53
0x18
```

I disconnected the only two things that were plugged to the Pi:
* a USB dongle for a wireless remote keyboard
* [raspberry Pi camera module with infrared leds](http://img.dxcdn.com/productimages/sku_390047_3.jpg)

And the beeping stopped. It nows replies 0x01 all the time, but why is this happening ? I mean why would more energy consumption from other peripherals would affect the UPS PIco ?

PiModules, 2017-01-21:
As I can see, PSU is good, or looks like good. However when you put all 
your hardware shows that you have significant voltage drop. I think that 
your additional hardware is drawing much more current time to time, so 
it cause the reported problem. We  have included in the ongoing firmware 
upgrade some extra mechanism that makes thy system less sensitive for 
such cases, however if you have voltage drops below 4.75V as defined on 
the USB specs, UPS PIco must take over and supply from battery. We can 
not change it much more, as then we will be not able to protect your system.

As I explained many times on out technical 
http://www.forum.pimodules.com/ forum, you have the following 
conditions, your extra hardware require continuously or time to time, 
more current, and this cause due to (again) bad PSU or powering cable 
voltage drops below the 4.75 V, this activate the PIco battery back up, 
after that in a few seconds PIco is detecting that you have connected 
cable power, and is switching to cable powering dis activating the 
battery back up. But due to low powering voltage, PIco again switches to 
battery backup and again, again again.
_
__You must use good PSU, and good micro USB powering cable. _
Try to connect one by one extra hardware to isolate which one is making 
your troubles , or select "true" 3A PSU @5.1V
In the mean time , keep you system for about 2-3 hours without extra 
hardware in order to charge the battery.
On my opinion the camera is doing that folks, check their current 
consumption

BR Ioannis @PiModules