modmypi/PiModules #5: Awful pcb quality

glorusso, 2017-04-05:
I'm writing this issue because my message on pimodules forum has not been published yet.
I have a UPS-PIco HV3.0A Plus and I'm not satisfied with this product at all.

First, given that this boards cost almost twice a raspberry PI3 I don't like to see smd resistors soldered like in the pictures below: 

[image]

[image]
[image]

PiModules, 2017-04-05:
> top-annotated-01
> side-smaller
> side-2-smaller

Hi,

Please provide me your nick name to see what you written on the 
pimodules forum. We are receiving so many sex oriented spam messages so 
may be it has been deleted by mistake. There are also rules than must be 
followed in the forum.

Now about your opinion.

1. The cost of these board is low, I think if you compare it with other 
boards doing less than this one. Regarding the compassion of this board 
price with the Raspberry Pi is wrong on my opinion. Especially as 
Raspberry Pi is manufactured in millions of units. Therefore it is not 
possible by nobody to manufacture such qty of any (our or other pcb for 
the Raspberry Pi). BTW there are many other boards with much higher 
price than Raspberry Pi, there are also bad ?

The quality of this PCB is on my opinion very good, it is 4 layers PCB, 
gold plated, with thermal pipes. Designed and simulated with very high 
tech CAD/CAM tools (www.pads.com)

Regarding these 2 resistors we added to the PCB, it is a functionality 
that has been invented after we manufactured the PCB (not planned at the 
beginning), however decided to add it, as we think that is very usefully 
for the users. These two resistors (by the way soldered very 
professionally under microscope to each boards separately) allows user 
to startup the system from EXP, even if battery is completely 
discharged. We called it cold wake-up, and it is very useful for 
industrial applications, and is requested by many industrial customers.

Regarding your "bad opinion about our product at all" you called it it 
would be nice to know point by point what you not like on it, so we will 
have opportunity to improved it.

We are receiving so many very positive comments on our product(s) so 
your bad opinion is very important to use, to see what we need to 
improve in order to make you also satisfied as all other users/customers.

Awaiting for your more detailed answer,

With my Warmest Regards

Ioannis

PiModules, 2017-04-05:
Hi,

Our product is very dynamically updated and new documentation and 
firmware is released every 3-4 weeks. We are releasing it on our forum, 
and when stable pass it to the GitHub.

The latest part of documentation is here

http://www.forum.pimodules.com/viewtopic.php?f=25&t=2898

There are thousands of various ideas, posts, questions and scripts that 
could be usefully for anybody.

Also it is supported on the ModMyPi on their forum

You can follow the installation guides from the GitHub

https://github.com/modmypi/PiModules/wiki/UPS-PIco-Installation

Therefore the simplest way to have support is to search the forum(s), 
GitHub or just ask for it
My Warmest Regards

-- 
Ioannis

www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

glorusso, 2017-04-05:
In my case, this kind of board is supposed to be used in remote locations where you need to trust both hardware and software against the odds of failure. Now... seeing such resistors soldered like that, however good the rework was, doesn't help.
What about documentation? Where can I find a comprehensive manual?

PiModules, 2017-04-05:
Hi,

Exactly for customers like you, we added these resistors. The cold wake 
up it very important on remote locations, so we decided after we 
manufactured this version of PCB, as many times battery can be 
completely discharged, and on such cases the cold wake up is necessary. 
Why do you think that these 2 resistors soldered very professionally are 
not so good as all others soldered by SMD machine ? As you can see on 
your photos they are soldered much better that the "original" ones and 
tested many many times before we released product to customers.

Please provide what are the points you called our product "not like at 
all" I would be very happy to improve it on each point you will mention 
about.

Please follow the installations guides in the GiHub, as also below links 
and you will have our product installed very easy.

There is a  cumulative manual, but due to rapid release of new 
firmware/features it has been not updated yet, so easier for you is to 
follow links I proposed to you

1. You can follow the installation guide from the GitHub

https://github.com/modmypi/PiModules/wiki/UPS-PIco-Installation

2. For the RTC on Jessie use the following info

http://www.forum.pimodules.com/viewtopic.php?f=26&t=3259

3. For the simple daemons installation use the following info

http://www.forum.pimodules.com/viewforum.php?f=26

4. Also this script

http://www.forum.pimodules.com/viewtopic.php?f=27&t=4870

I would kindly ask you to move this discussion to one of our technical 
forums

Warmest Regards

-- 
Ioannis
www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

glorusso, 2017-04-05:
About documentation here's my opinion:

1. A person who gets this board for the first time would be **extremely** confused about your documentation. You can't tell people to go and look on the forum for simple issues. There needs to be a well documented set of `use cases` like: 
  - `UPS with automatic shutdown after 30 s`
  - `UPS with external power supply, shutdown after 30s`
  - `UPS with ..., + external relay management`

2. Also you know that most people do not remember more that 3 to 5 things at the same time... so why on earth did you choose to use ***9 led** with pulse rate coding?... I mean it's got to be simpler.

3. A one file manual is mandatory for this kind of product. Take  for example the [LifePO4ered-PI manual](http://lifepo4wered.com/files/LiFePO4wered-Pi3-Product-Brief.pdf)