modmypi/PiModules #7: 120% of battery percentage?

grzegorzshub, 2017-04-29:
I have got a UPS PIco HV3.0 A Stack 450 with default 450mAh LiPO Battery. 
Status script shows 120% of battery percentage. Any clue why it is not 100%? Report below:
***********************************
      UPS PIco HV3.0A Status       
***********************************
 
  UPS PIco Firmware.....: 31
  UPS PIco Bootloader...: 53
  UPS PIco PCB Version..: 41
  UPS PIco BAT Version..: LiPO (ASCII: S)
  UPS PIco BAT Runtime..: 1 MIN
  UPS PIco r232 State...: OFF
 
  Powering Mode.........: RPi POWERED
  BAT Percentage........: 120.0 %
  BAT Voltage...........: 4.3 V
  RPi Voltage...........: 4.9 V
  NTC1 Temperature......: 38 C
  TO-92 Temperature.....: 12 C
  Extended Voltage......: 0.0 V
  A/D2 Voltage..........: 0.0 V
 
  PIco FAN Mode.........: ERROR
  PIco FAN State........: OFF
  PIco FAN Speed........: 320.0 RPM
 
***********************************
         Powered by PiCo           
***********************************

LmV4corbeau, 2017-09-07:
I have the same Issue:
UPS PIco HV3.0 A Plus Stack 450 with default 450mAh LiPO Battery.

***********************************
      UPS PIco HV3.0A Status       
***********************************
 
  UPS PIco Firmware.....: 35
  UPS PIco Bootloader...: 50
  UPS PIco PCB Version..: 41
  UPS PIco BAT Version..: LiPO (ASCII: P)
  UPS PIco BAT Runtime..: 128 MIN
  UPS PIco r232 State...: OFF
 
  Powering Mode.........: RPi POWERED
  BAT Percentage........: 120.0 %
  BAT Voltage...........: 4.3 V
  RPi Voltage...........: 5.0 V
  NTC1 Temperature......: 36 C
  TO-92 Temperature.....: 15 C
  Extended Voltage......: 0.0 V
  A/D2 Voltage..........: 0.0 V
 
  PIco FAN Mode.........: ERROR
  PIco FAN State........: OFF
  PIco FAN Speed........: 320.0 RPM
 
***********************************
         Powered by PiCo           
***********************************

xpavli44, 2018-01-18:
Same problem here, also   
TO-92 Temperature.....: 11 C does not seem right

***********************************
      UPS PIco HV3.0A Status
***********************************

  UPS PIco Firmware.....: 35
  UPS PIco Bootloader...: 53
  UPS PIco PCB Version..: 41
  UPS PIco BAT Version..: LiPO (ASCII: S)
  UPS PIco BAT Runtime..: 1 MIN
  UPS PIco r232 State...: OFF

  Powering Mode.........: RPi POWERED
  BAT Percentage........: 120.0 %
  BAT Voltage...........: 4.3 V
  RPi Voltage...........: 5.0 V
  NTC1 Temperature......: 46 C
  TO-92 Temperature.....: 11 C
  Extended Voltage......: 13.3 V
  A/D2 Voltage..........: 0.0 V

  PIco FAN Mode.........: ERROR
  PIco FAN State........: OFF
  PIco FAN Speed........: 320.0 RPM

***********************************
         Powered by PiCo
***********************************

The-Sleepy-Penguin, 2018-01-24:
I also have the same issue - when the battery is charging, I get:
`BAT Percentage........: 120.0 %`
I am using 18650 batteries.

The-Sleepy-Penguin, 2018-01-26:
@grzegorzshub @LmV4corbeau @xpavli44 
Hello all,
I think I might know why we are having this issue - as the UPS Pico HAT is designed to work with 3.7v batteries, these *usually* have a maximum voltage of 4.2v - this is what the HAT is expecting. From what my HAT is telling me (and by the looks of it yours as well) it shows that your battery is running at 4.3v. Now, it looks like this is caused by incorrect/rounded readings by the HAT when the batteries are charging (all of your HATs say that your RPi is powered from the mains when you get 120%), and so when you do the maths, a voltage of 4.3v will give you the 120% charge level.
I hope that this has cleared things up!

xpavli44, 2018-01-28:
@The-Sleepy-Penguin thank you for the clarification, the sad thing is that I use battery which came with the Pico HAT and I would not expect this to happen :( with original parts

PiModules, 2018-01-28:
Hi,

The A/D converters (all 3 of them implemented in the UPS PIco) are using 
resistors dividers in order to adjust the voltage levels to the micro 
controller A/D converter range.

This is true for any other A/D converter using in any PCB that use the 
original micro controller embedded A/Ds.

In addition we are using the internal (micro controller one) voltage 
reference.

The resistors divider is based on 1% accuracy resistors, similar the 
internal voltage reference his accuracy is about 1-2%.

Putting all these things together, you will receive the result accuracy 
about 2-3% of measure. Therefore readings will vary from PIco to PIco 
about 2-3% due to above mentioned facts.

This is valid to any PCB (not only PIco) that uses

How it can be resolved?

In the conversion part of firmware (this one that is converting raw A/D 
data in to Voltage readings) we use a dedicated factor. However this 
factor is calculated to exact resistors values (without taking into 
account the 1% accuracy of them) as also the exact value of the internal 
voltage reference (without taking into account the 1-2% accuracy of it).

Therefore this factor need to be calculated exactly as it is on each PCB 
(in reality is is not possible), however we are implementing the 
following algorithm.

1. As the LiPO charger IC is very accurate and always when battery is 
full charger it is exact 4.2V (with accuracy better than 0.1%)

2. We are implementing adjuster of the A/D converter that will takes 
this as reference (the charged battery level)  and slightly change the 
battery A/D converter factor providing the proper micro adjustment of it.

However I would like to note (as a designer of PIco Hardware and 
firmware) that on my opinion it is not necessary, as all read A/D data 
are passed via Olympic Score filtering, Low Pass software Filtering as 
also via rolling mid value algorithm (with sliding window of 16 
readings). All that added values make sure stable and repeatable A/D 
readings. The end values is also rounded on the first position after 
comma. So, if the reading is 4.22V it is shown as 4.2V, if it is 4.25V 
it is shown as 4.3V etc. However please kindly notice that it is not a 
multi-meter and just and very good measure (but not multi-meter)

With the new ongoing firmware we have implemented this mentioning above, 
micro adjustment algorithm for battery reading A/D, and all others A/D 
converters.

My Warmest Regards

Ioannis/PiModules

Serveurperso, 2018-10-13:
WTF ?!? Never direct convert a voltage to a bat percent !!!! you must use current integration + auto-calibration algorithm to get the a real bat % like any smartphone or laptop !!!!

PiModules, 2018-10-13:
Hi,

I know it. This script is not written by our company, one of users wrote 
it and offer it for free (thank him very much for that), you are free to 
write your own if not like that and share with others like he done

Kind Regards

Serveurperso, 2018-10-13:
Yes this script is very usefull as sample for i2c commands.
I say this because the hardware need to integrate mA and give a % value inside a register. It's an usefull and basic feature for a battery management system inside robot or any battery operated project