modmypi/PiModules #8: Feature Request: Battery runtime before shutdown < 60 seconds

svenheitmann, 2017-05-17:
Hi everybody!

With the UPS PIco HV3.0A you are only able to set a battery runtime of at least 60 seconds (writing 0x00 to the 0x01 address on 0x6b according to your documentation) resulting in a battery runtime of 60 + 0 * 60 seconds, thus limiting the user to a minimum battery runtime of 60 seconds.

Anyway, for some applications it's useful to have an even shorter (= below one minute) battery runtime before the ups sends the shutdown signal. An option for an immediate shutdown could be useful as well.

PiModules, 2017-05-17:
Hi,

Yes, this time has been fixed to minimum 60 seconds. It is for many 
reason, usually this time is minimum needed to have system working on 
battery. Also, some times, mainly due to bad quality of PSU, system can 
switch to battery, and back, so we need to "filter" such cases. I think 
that 60 seconds is logical time, would not like to change it to shorter.

However if you really need to have it shorter, then you will need to 
monitor when system is in battery mode (0x00) and if wait little bit 
about 10 seconds and initiate the shutdown manually (by using python 
script).

by simple "sudo halt" or sending command to the PIco that start 
immediate shutdown. Both cases are good and recognized by PIco

In the next version of  firmware we add interrupts that will inform you 
(without polling) about changes of selected registers, but not now

BR  Io

svenheitmann, 2017-05-18:
Hi Io,
thanks for the quick reply. Although I would be, in general, quite careful with assumptions about your customers' needs when it comes to the level of granularity that you expose for controlling electronic components, I see your point.
I followed your advice and simply initiate the shutdown process after the system has switched to battery mode for a certain amount of time. It works as expected. 👍 

I am looking forward to the next version of the firmware! :-)

Best,
Sven

tanabi, 2018-09-09:
@svenheitmann  Hey there :)  Could you provide your script as a sample?  I want to implement this myself.