modmypi/PiModules #11: Pico UPS HV3.0A does not use eternal power

KarolisMil, 2017-09-12:
Hello,
What issues may cause that Pico UPS does not use external power when it is present? When I start the device by plugging battery in and then applying external power, it boots up and starts RPi. After couple of seconds on my lab bench power supply (set to 5V, 3A) I can see, that no power is used. At the same time LED blinks slower and I2C responses about battery power mode. After about 60 seconds system is being turned off by Pico UPS.

P. S. I assure you that my power supply good and it cannot be an issue here.

kessero, 2018-02-01:
Did you solve the problem? I've something similar. My pi works on power and when I turn off power it works on UPS with battery 4000 mAh. After about 90-110 seconds system turn off it looks like ups get info that battery is low and shutdown the system. Sometimes it's turn off some times red Users led stay ON,  on pi ups and i have tu press button UR to complete shutdown electronic. With'out pressing button UR when I plugin power back it's not starting raspberry. So i have two problems with this Pi hat 3.0:
- one big battery 4000mAh and very fast shutdown 
-  two it's not starting back after power is back with'out pressing the UR button?

PiModules, 2018-02-01:
Hi,

There could be one of the below described reasons for your mentioned problem

1. Normally the PIco HV3.0 is set (by default) to be working on battery 
for 70 seconds. It is independent form what battery is used. If you use 
bigger battery you should change this to requested running time i.e. 10 
minutes or 50 minutes, or if you like to unlimited time (then system 
will safety shutdown when battery is near to be empty 3.5V for LiPO and 
Li-Ion batteries). So the most possible reason is the you have not set 
it properly and it is still the default value

2. The second possible reason is that you have not setup properly the 
daemons (needs to be set-up both), so PIco is assuming that Raspberry Pi 
is not running, so after pre-defined time cut the power. You should 
check the UPS LED is blinking after boot-up of Raspberry Pi (faster when 
cable powered, and slower when battery powered) to be sure that Daemons 
are installed properly and interacting with PIco pulse train.

It is about 30 seconds continuously light of UPS LED (booting time of 
the raspberry Pi), the UPS is blinking fast (as far you is powering the 
system with cable power), the (if cable power cut) UPS LED is blinking 
much slower, and if you press the FSSD button, battery is low,  or 
battery running time is defined to some value, UPS LED is again 
continuously until system shutdown. When system shutdown, PIco is 
waiting longer time (UPS LED lit) waiting for possible reboot of the 
Raspberry Pi.

It is mandatory to have installed the Gold Plated Reset Pin on your PIco.

Please describe in bullets your problem so it will be much easier to 
answer and fix your problem

 From what you are describing, seems that you have not properly 
installed the Daemons, so please follow step by step the installation 
procedure described in the manual or use one of the automatic 
installations scripts available on our unofficial but very active 
technical forum

Please do not hesitate to come back with questions if you still have a 
problems
BR Io

kessero, 2018-02-01:
I can't find in manual this procedure
_

> 1. Normally the PIco HV3.0 is set (by default) to be working on battery
> for 70 seconds. It is independent form what battery is used. If you use
> bigger battery you should change this to requested [...]

_
How can I change this value? I think this could be my problem.
UPS led slower blinking when works on battery.
Do I really need this Gold reset pin I've raspberry 2?

PiModules, 2018-02-02:
Load this manual, and check page number

www.pimodules.com/_pdf/15_0x40_W_UPS_PIco_HV3.0.pdf

PAGE 137

bat_run_time register,

usage

sudo i2cset -y 1 0x6B 0x01 0xff  - for unlimited running time (until 
battery is empty)

sudo i2cset -y 1 0x6B 0x01 0x01  - 1 minute + 10 seconds (total time is 
70 seconds), default value

sudo i2cset -y 1 0x6B 0x01 0x02  - 2 minutes + 10 seconds (total time is 
130 seconds) etc

It must be better described in the manual I will do it soon

BR Io

Prospectus for the new PIco HV3.0_B_ are here:

http://www.pimodules.com/_pdf/UPS_PIco_HV3.0B_HAT_Stack_TopEnd_Prospectus.pdf

http://www.pimodules.com/_pdf/UPS_PIco_HV3.0B_HAT_Plus_Prospectus.pdf

kessero, 2018-02-02:
Thanks for your help I found this `sudo i2cset -y 1 0x6B 0x01 0xff` on this forum [](http://www.forum.pimodules.com/viewforum.php?f=20&start=50).
Now seems to work OK.