modmypi/PiModules #15: Allow using alternate GPIO pins or eliminate gpio consumption with i2c based gpio

Dloranger, 2018-03-11:
Common to many boards of this nature, it would be great to be able to use alternate GPIO pins to avoid a conflict with other HATS.

I am building a fairly complicated system and find that the two gpio you used are already in use by other boards.   I would like to have the Pico UPS designed into the system as the usage may be far up in the mountains where it is difficult to recover from a bad shutdown, right now I can't do this.

Short term - what changes can I make to the board and software to get the signals onto different GPIO pins?  I am very comfortable with cutting traces, and doing very fine pitch soldering if appropriate.  I  have access to great tools so I can do most any reasonable mods.  I would like to use pins 11 and 17 currently as those are still open and most convenient.

long term - Unless there is a strong reason against it, you are already consuming the I2C bus, so why not use an expander like the MCP23017 or similar to route the GPIO back into the processor?

PiModules, 2018-03-11:
Hi,

This PCB is 4 layers, 6/8 mils density. It is not possible to do any 
changes by hands.

We plan to have version of firmware that is not using any GPIO. However 
due to unexpected changes in production we ware forced to change our 
plans for some weeks. We will announce a very good news this week, by 
the end, that will cover your requirements. Please be patient as we are 
working hard to do the best possible.

Also due to density of components we used on this PCB, it is ultra 
difficult to add any other IC without changing of the technology of PCB 
(smaller vias, smaller components - smaller than 0603, or more layers) 
that cause an increase of price, and new production setup, facts that we 
would prefer to avoid as it would be generating a strong delays in 
delivery again.

Please be patient, solution we will offer will  free the GPIOs.

There is no place to put even jumpers (without penalties), as each place 
of the copper has been used for cooling, high power tracks, or just 
signal tracks.

We are working on GPIO version of firmware compatible with all version 
of PIco A,B and any other new one.

Best Regards

Io

Dloranger, 2018-03-11:
Good to hear there is a firmware update coming very soon that will fit my
needs, many thanks!

I won't see my hardware for about 2 weeks, so any time before then would be
fine.

By The way, The board is highly recommended within my contacts. Good work.

Dloranger, 2018-03-21:
Any updates on the update?  I got my hardware delivered today.  Thanks

Dloranger, 2018-03-29:
Any updates on this issue?

Dloranger, 2018-04-04:
Any updates on this issue?  It has been 3 weeks since you said "We will announce a very good news this week, by the end, that will cover your requirements. "  Still waiting...

PiModules, 2018-04-04:
Hi,

We have had some production (of the HV3.0B+) issues that required to be 
urgent solved, we are under final testing and just after the Easter (our 
Orthodox Easter, we have it one week after yours) will be released

Warmest Regards

Ioannis

Dloranger, 2018-04-22:
Any update on the new code, it has now been 18 days since you promised an updated code would be released.