modmypi/PiModules #16: Documentation for extra parts

Surfking55, 2018-03-17:
I just received my PIco and started looking through the install / setup documentation and can't seem to find anything on these extra parts that came with my kit. 

Two pin header w/ jumper - where does this go and is it needed?

Relay - Loads of questions about this one, I'll just leave it at that

On / Off (I'm assuming) switch - Where does it go and is it needed?

Single long pin - Where does it go and is it needed?

Foam sticker - Where does it go and is it needed?

Thanks
[image]

Dloranger, 2018-03-21:
no documentation included for me either.  

most of the parts are intuitive, but still nice to have 1 sheet of paper included with the instructions just to be sure we are doing it right.

PiModules, 2018-03-21:
Hi,

I will update the latest version of manual. I'm very sorry for that 
delay, but the release of the new Raspberry Pi Model B+ changed all our 
planning we have had, as we ware obligated to change production to the 
new Model of UPS PIco HV3.0 B+

I will update this files by tomorrow

My Warmest Regards

Ioannis

Surfking55, 2018-03-22:
Also, is the forum.pimodules.com site still working / used? I have tried twice to post a question and it never shows up.

xoul, 2018-07-20:
Me too I'm interested in the two pin header w/ jumper: is this just needed to write-protect the EEPROM?

xoul, 2018-07-20:
Again, on the two pin header with jumper, I can confirm it is used to write-protect the EEPROM.
See https://github.com/modmypi/PiModules/issues/20.

PiModules, 2018-07-20:
Hi,

In the Package of the UPS PIco Hv3.0B there is one jumper 1.27mm and it 
is used to make possible to write the HAT EEPROM

It is not necessary to solder it, only if you like to use this EEPROM.

In the Package of the UPS PIco Hv3.0B+ there is one jumper 1.27mm and it 
is used to make possible to write the HAT EEPROM

It is not necessary to solder it, only if you like to use this EEPROM.

But in the Package of the UPS PIco Hv3.0B+ there is also a second jumper 
2mm. It is used only when UPS PIco Hv3.0B+ is assembled for the 
Raspberry Pi 3 Model B

A detailed description about it is provided in the latest manual here

https://drive.google.com/open?id=1YkKdjYMnC6vPaf1LdNnbcgyqQ3CAraJJ

Please find attached the latest NOOBs (8GB) with everything installed,

The User is : pi

The Password is : 1

https://drive.google.com/open?id=1GT3Q0WAXriDvFNevW2MRNGomZwrixBxI

PiModules, 2018-07-20:
Answered already on my previous post

PS. We have had a strong problem with our (very active technical support 
forum - www.forum.pimodules.com ). Problems (huge number of spams)  
solved and forum will be back and running by tomorrow. There are huge 
number of questions and answered about UPS PIco