modmypi/PiModules #19: Power through GPIO - UPS blink,no power information

kessero, 2018-03-24:
In my project I've to connect more PSU. Rapberry has some fuse with limit power too 1,1A. I need more.
But if I connect PSU through GPIO pins. UPS pico blink UPS led and sudo i2cget -y 1 0x690x00 return 0x02.
I need to have info when power is from BAT or PSU but PSU must bi connected to GPIO or some how but not through raspberry mico usb?

PiModules, 2018-03-25:
Hi,

If you need to connect PSU directly to GPIO you need to take care for 
the reverse powering, so need to have between 5V GPIO and your PSU 
shotky diode or better ZVD circuit as it is in the Raspberry PI powering 
path. If you use shotky diode, your PSU need to provide 5.2-5.3V, as you 
will have voltage drop. The rest of your question need to be better 
explained, please do it

Warmest Regards
Ioannis
www.pimodules.com

-

kessero, 2018-03-26:
OK, now I know that I have UPS without Plus. So basically I can't have external PSU and working correct UPS. In my project I need to know when power is from PSU or BAT. When I connect external power to GPIO pin's, on UPS blinking only UPS led and device inform me that is working on BAT. For run I've to connect for a moment power to microUSB. For now I made some modification on raspberry to have some more power from PSU thru microUSB port and some problems are off. I need about 1,6-1,8A with is more than raspberry can pass(1,1 fuse on microUSB port). Next time I've to search UPS Pico with Plus with give me possibility to connect external PSU.

PiModules, 2018-03-26:
Hi,

The slide switch is working as a battery ON/OFF on your version, 
therefore you can use it as a power bank if I understand you properly

BR Io