modmypi/PiModules #21: configure magic-switch

nomad-fr, 2018-07-18:
Hi, I just have read the new manual version .
```
3.0 15/06/2018 all Updated to Version HV3.0B/C with new
functionalities and B+ assembly Instructions
```
I still don't understand which register I have to set to which value to enable magic-switch :

- power off of pi when slide is in off position (with battery and don't care about of Cable power)
- power on of pi when slide is in on position (with battery and don't care about of Cable power)

There is some new information on page 84, but it's not really clear for me.

PiModules, 2018-07-20:
Hi,

Please note that UPS PIco firmware recognize on what model is running 
(HV3.0A, B or B+) automatically based on bootloader info. Therefore some 
functionalities are automatically activated with specific implementation 
to each model.

The Magic Switch is not activated in Model UPS PIco HV3.0B+, and is 
activated only on Models UPS PIco HV3.0B. I will activate it in the 
firmware also for model UPS PIco HV3.0B+ (that has slightly different 
functionality) we are going to release on Monday/Tuesday with this and 
others functionalities

---

Please do the following in order to activate the Magic Switch fro UPS 
PIco HV3.0B

With Cable powering connected do the following

1. Make sure that Magic Switch is on Position ON (on the LEDs side)

2. Check the Magic Switch Register

*i2cget -y 1 0x6b 0x16*

should be 0x00 or 0xff

3. Then program it

*i2cset -y 1 0x6b 0x16 0xaa *

4. Remove Powering Cable

5. It is still running, but on battery mode

6. Switch the Magic Switch to OFF position

7. System start shutdown, and after short time cut the power

8. Whenever you like can make it ON/OFF (of is always with system 
shutdown first)

IMPORTANT NOTICE: When the Magic Switch Register is programmed, the 
battery running register is automatically set to 0xff, so system is 
running on battery until it is low, then system automatically shutdown. 
If you like to have it running shorter, you need to reprogram the 
register for a shorter time with just after Magic Switch programming

f0k, 2019-07-19:
> The Magic Switch is not activated in Model UPS PIco HV3.0B+, and is
activated only on Models UPS PIco HV3.0B.

Has this changed in between? Do I need to do anything special to have the magic switch shut down the Pi when switched off, and power it on when switched on, no matter if running on power or battery? (It's for a remote site, I can instruct people there to turn off and on a switch if it stopped working, but not to connect to the Pi via ethernet and SSH.) On the first try, after installing the software, the switch seemed to have no function.

PiModules, 2019-07-21:
Hi,

Yes, you need to_*program this switch to be usable*_ on both models as 
firmware is common.

Please read the manual it quite good described, and based on the manual 
ask for any clarifications

When you switch OFF with Magic switch the system (RPi + PIco) they it 
takes care to properly shutdown the Raspberry pi and then cut the power. 
So, you are completely safe.

You need SSH only for programming or terminal, then it is working 
automatically. It is working only when system is powered by battery. For 
the Raspberry Pi B+, and proper PIco version (if you have installed the 
PE pin) then it is working also when cable power is used.

BR Io

f0k, 2019-07-25:
> Hi, Yes, you need to_*program this switch to be usable*_ on both models as firmware is common. Please read the manual it quite good described, and based on the manual ask for any clarifications

Yes, I tried that before, but wasn't sure if the manual was complete. I tried to download the latest manual by following the link "Latest Firmware and Manual" on [the product page](https://pimodules.com/product/ups-pico-hv3-0b-plus-hat-stack-450-poe), but it's a 404. I found a manual from 15.06.2018. It has a chapter on the magic switch with an incomplete table on the powering behavior depending on the switch state and some "PICo registers", but it always says "Standard FSSD functionality and Low Powering as described in other chapters". There were no instructions on what to install.

I now found a later manual via Our Products -> UPS Pico HV 3.0 A/B/B+ -> [Common Updates](https://pimodules.com/common-updates), in the form of a news entry: https://drive.google.com/open?id=1-qsnRpp6tYCa9hC7_SYldHXyOm-1pYLi. It explains well what to do!

> For the Raspberry Pi B+, and proper PIco version (if you have installed the PE pin) then it is working also when cable power is used.

Great! One question though: When I do `i2cset -y 1 0x6b 0x16 0xaa` while the switch is in OFF position, will the system shut down immediately? I am not sure what position I left the switch for one of the remote devices. Thanks!

alijem, 2019-08-07:
Hi,

I tried to use the magic switch with a ups pico hv3.0b+ HAT Model Stack 450 (fw 0xe3) with a rpi model 3b+. **I wasn't able to make it work.**

First, I soldered the gold plated pins to the P4 and PE based on the instruction in _A1ssembly of the Gold-Plated Hardware Reset Pin (POGO Pin)_ from the manual. 

Second, I modified the register based on the instruction in _Setting-up the Magic Switch_ from the manual. 

But, switching on/off the magic switch make no difference. The raspberry never turn off.

Still the hard reset work when pressing on RR. 

@f0k Did the magic switch worked for you ?

@PiModules What did i miss ? Do you have any fix to make it work ?

Thanks for the answers.

f0k, 2019-08-20:
> @f0k Did the magic switch worked for you ?

I couldn't test it myself, but colleagues said it works...

alijem, 2019-08-24:
> > @f0k Did the magic switch worked for you ?
> 
> I couldn't test it myself, but colleagues said it works...

Hi,

@f0k  Thanks for your reply.

I tried with two rpi and two module. It didn't work. 

So i came to use ssd button to start/stop the rpi. But it will be much cleaner to use the **magic switch**.

PiModules, 2019-08-24:
Hi,

I checked it million times and it is working. Please report firmware 
version to check again it

My be 1 per million I done something mistake on the firmware

Please report it, to check as we have ready newer firmware to release

Please follow exactly instruction form manual, please exactly

BR Io