modmypi/PiModules #26: Battery not holding charge

carcery, 2018-09-14:
UPS Pico HV3.0B+ with 450mAh LiPO. on a RPi 3B
While I'm utilizing the system, with power cable and battery connected, the battery drains and the system starts beeping. 

I have assembled the hardware and installed the software according to the instructions (minus RTC). 

Thanks

PiModules, 2018-09-15:
Hi,

You have a weak PSU, that means when PIco is switching battery charger 
on, it causes voltage drop, that is interpreted by the PIco as cable 
power loose. So, battery charger is switched off, that cause after some 
seconds stable powering, so system is trying to switch battery charger 
again, and loops continue, until battery discharge

You need much better PSU (recommended 5.2V @3A) with good micro USB 
powering cable, recommended 22AWG

BR Ioannis

*-- *

*Ioannis****www.pimodules.com****--*

carcery, 2018-09-16:
I don't think that's the case, I have this system connected to a 5V 5A power supply. Maybe I have a bad battery. I have reinstalled the whole think a couple of times and I am getting some results

carcery, 2018-09-16:
Also, I have tried to charge the battery while the pi is off, but it's not charging.  While the pi is off I have the following lights:

UPS ON SOLID
BAT ON SOLID
CHG OFF

PiModules, 2018-09-17:
Hi,

Please advise what system do you have. What model of PIco do you have. 
How are you powering it, via micro USB or via EPR ?
Also please advise if your USB LED is blinking and how fast
Regards

carcery, 2018-09-17:
UPS Pico HV3.0B+ with 450mAh LiPO.

Left the system off but connected (to charge the battery) for an hour. The battery is not charging. The battery is bad

PiModules, 2018-09-17:
Again, What model Stack or Advanced ?

What do you mean "Left the system off"

Please answer to my questions is the UPS LED blinking or not ? and if so 
how fast

the best is to send a short video and some pictures of you setup

carcery, 2018-09-17:
450 Stack model. What I meant was I turned of the Pi via menu>shutdoen but left everything connected.

You previously asked about USB light. Not UPS, which I answered. 

And in my previous post I stated that UPS light stayed on while shutdown. While the system is ON the UPS light blinks 400/400.

PiModules, 2018-09-17:
in the *text*

>
> 450 Stack model. What I meant was I turned of the Pi via menu>shutdoen 
> but left everything connected.
*To charge battery in current version of firmware Raspberry Pi must be 
running, as there are many other parameters checking then*
>
> You previously asked about USB light. Not UPS, which I answered.
*I asked if the UPS LED is blinking*
>
> And in my previous post I stated that UPS light stayed on while 
> shutdown. While the system is ON the UPS light blinks 400/400.
*That means you have a weak powering cable, please send pictures of your 
system
for 99.9% battery is OK,
*

carcery, 2018-09-17:
"Also please advise if your USB LED is blinking and how fast Regards" that's what you said previously. Just look up.

My cable is 16AWG coming out of a DC step down from 12V to 5V, fused with 5A. Not a weak cable.

Battery reads 0.0 on my voltmeter.

PiModules, 2018-09-17:
maybe cable is not contacting well, cable 16AWG is not weak, you are 
right, however how you have be able to connected such cable to micro USB 
connector ? May be connectivity of them is bad, we have here AWG22 
connected, and I think this is the maximum able to be connected to micro 
USB cable. You bought this cable? or done yourself ?

Please send me pictures of your battery, battery connector, and micro 
USB connector

alternatively you can for testing purposes only power your system via 
GPIO 5V (be very carefully with polarity), and then see if battery is 
charged.

Also if battery cable is damaged you should have very fast blinking CHG 
LED (very very fast), is that happens? it must be if battery is bad

It would be very helpfully if you can send me pictures I requested as 
also if possible a small video

carcery, 2018-09-21:
Due to your terrible support I decided to go with a different system. 

Guess what, it's working great, no issues with the wiring. It has been working great for almost a whole week. And it is recharging the battery without any problems.

I want to return your device and get a refund. How do I start the process.

PiModules, 2018-09-21:
Hi,

We have set you many questions asking you to answer to them. Each your 
question has been answered.

However, provided bu you answers now covered the my questions

1. It is not possible to have cable 16 AWG connected to micro USB 
socket, it is physically not possible to be made.

2. Due to that reason, I kindly asked you to provide me photos of your 
installation or a short video showing me how your system is working, not 
received until  now

3. As we have not received any certain answer to our question we ware 
not able to help you.

4. As physically it is not possible to have such connection of 16 AWG 
cable, I assume that the reasons of request of refund of the system is 
different than the technical issues, if so

5. I was asking you many times, to provide me the following info, photos 
or even better a short video, that will be solved any misunderstanding

Please limit your comments to technical issues only, as it is technical 
support forum only, and if required refund please address to 
support@pimodules.com

Then I will check your order

My Warmest Regards