modmypi/PiModules #28: Can't use Basic Schedulers and Events Triggered RTC based System Action Schedulers

daghemo, 2018-09-18:
I know that the UPS PIco HV3.0 has implemented 2 independent, Time Schedulers. There are:

- the Basic Time Scheduler (BS)
- the Event Triggered RTC Based System Actions Scheduler (ETR SAS)

They both seem to have been implement around 15/07/2017 between firmware 0x39 and 0x40.
The above information have been take from what the last version of the user guide printed on 05/09/2018. Most important, the product pages on the Pi Modules shop seems to confirm their presence.

It seems that both the BS and the ETR SAS can't be used right now, nor they could have been used since the beginning, if I'm not wrong. I really want/need these features. Don't know if other customers agree. I just write an issue on the subject in order to receive updates.

daghemo, 2018-09-28:
@PiModules can you write any update on this here?

PiModules, 2018-09-28:
as I wrote you to personal email, was planned to be released yesterday, 
but due to overload and need to make more tests, will be released today 
or tomorrow evening. We changed it little bit (improved)

daghemo, 2018-09-28:
I try to change the question a little bit...
Considering that the forum seems to have some problems right now, considering that the news of your blog/site do not contain any information about what has been fixed/added, considering that the changelog within the user guide has not been updated, considering that It's not the first time you tell me that this would be implemented for real very soon, could you please write here when the BS has been implemented, tested and officially released (no Beta) and when the ETR SAS has been implemented, tested and officially released (no Beta)?

daghemo, 2018-10-04:
Broken link on http://pimodules.com/products/ups-pico-hv3-0a-b-b/common-updates.

lz2plm, 2019-04-29:
I am also looking for more information of how to use the Event Triggered RTC. Is it possible to start the RPi, based on the voltage level of the battery? 

Thanks

daghemo, 2019-05-04:
@lz2plm, I don't know if the Event Triggered RTC has been finally implemented, but I'm know using other UPS HATs for the Raspberry Pi.
I'm curious: why would you like to start the RPi based on the voltage of the battery?
That is, if the battery level is low, it may be not safe to start the RPi, and if the battery level is high, well, maybe there is some kind of power supply attached and it makes sense to use it.
Maybe you are running of battery only and want to start the RPi al long as someone replace the battery with a good one?