modmypi/PiModules #29: Bi-Stable relay not working on HV3.0A Stack plus

jvillagomez, 2018-09-20:
i ordered a PCB with relay installed already. I can hear the switch go back and forth when sending the commands, but there seems to be no contact going through. 

Commands used are:

`
sudo i2cset -y 1 0x6b 0x0c 0x01 (and 0x00)
`
Has anyone else had this problem?

PiModules, 2018-09-21:
Hi,

What model of PIco do you have ? normally you should have a short click 
when switching. I confirm your command, it is OK.

As you ordered the PIco soldered, it has been ordered from us. Can you 
please send me a photo of the bottom side of the PIco to see if Relay 
has been soldered properly ?

jvillagomez, 2018-09-22:
I am using the "" model.
I can hear the clicking when I switch using the command mentioned above, but nothing happens. I believe no contact is occurring inside the relay block. This happens whether I power the pi setup via usb power (on the pi) or with the LiPO4 battery pack.

I have attached a photo of the setup I tested with the relay, and several photos of the soldering.

[image]
[image]
[image]
[image]
[image]

PiModules, 2018-09-22:
> 42300481_1915709048508434_4003906815573098496_n
> 42304789_966959893512506_2504491908147970048_n
> 42310687_575938452820327_5084550604773130240_n
> 42446230_1157397394414957_1455216812026232832_n
> diagram

Hi,

You have the UPS PIco HV3.0A Plus

Your latching relay is not fully soldered. If it has been done by our 
company, give me the order number so I will cut hands one of our workers ;-)

If you done it yourself, you need to solder all pins of the latching 
relay (and save hands of our workers)

I would also kindly ask you to show me with additional photo of the 
other side of PCB to be sure that everything are OK.

As you can see some pins of it are not soldered, so you need to have it 
solder them

I can not understand your diagram, please describe what are you going to 
do in order to avoid something wrong (I have this suspicious)

Kind Regards

jvillagomez, 2018-09-22:
Yes, it was one by your company. I don't have the order number right now, but will look for it and follow up with you.

Thank you for your help! I will solder the remaining pins and test again.

Sorry, in the diagram the "regulator" is just a load. I will test the same setup after soldering the pins.

Thank you for the quick reply.

[image]
[image]
[image]

PiModules, 2018-09-22:
> 42304068_1008270046002762_3653192031133499392_n
> 42344880_241100979887985_5715974306966536192_n
> 42402400_621674058228244_6271671542388621312_n

One young man, just loosed his hands ;-)

Unacceptable for our side. I'm very sorry for that

Please solder the other pins, and everything will be fine.

The latching relay is placed properly, so just solder the other pins

Remember that for HV3.0A, the relay is rated only to 1A max !!, for 
HV3.0B there are two places and one is rated 2A, and for HV3.0B+ is is 
rated for 2A