modmypi/PiModules #32: Battery discharging too quickly

jvillagomez, 2018-10-06:
I bought the 8000mAh battery hoping to get long hours out of it, but it discharges too quickly. I only get about 90 minutes out of it before FSSD (3.6V -> 3.0V).,

The pico board is set to the correct profile and we charge it with 5V 3A charger.

Board:
UPS Pico HV3.0A Stack Plus

Battery:
LiFePO4 3.2V
Capacity: 8000 mAh

[image]

PiModules, 2018-10-08:
> discharge_case4

Hi,

It is depending what current you are drawing form the system on battery 
powering.

It should be running on 3V long time as the 3V is nominal voltage for 
this battery chemistry, it is very flat on this value

Please leave it until system automatically shut down, due to low battery

Please do the calculations as described in the manual, PAGE 96*
**
**https://drive.google.com/file/d/1-qsnRpp6tYCa9hC7_SYldHXyOm-1pYLi/view*

*
*

jvillagomez, 2018-10-19:
Hi PiModules,
The order number for my Pico and 8000mAh battery is **#206153747055**