modmypi/PiModules #35: undervoltage detected when PIco boots

gsemet, 2018-11-18:
Hello.
I have a bunch of Rpi  and PIco UPS to protect them, and some have undervoltage issue upon boot:
- Rpi3b + PIco UPS for mediacenter. So far, never had any issue
- RPi3b+ + PIco + 2.13in epaper screen for PiHole: when booted up, soon I get an "undervoltage log" and the rpi shutdowns after 40 seconds. To have a stable boot I need to:
  - plug the power. The Rpi starts
  - when I get the undervoltage error, unplug the 5V cable.
  - wait for a few seconds, plug again, boom, the status script says "Rpi powered", and could stay like this for weeks
- Rpi3b+ + PIco + USB keys (zwave sticks, rfxcom) for domoticz. Same issue, undervoltage log and auto-shutdown after 40 sec. The previous tricks does not work, I have to plug a self-powered USB hub, **without** anything connected to it. Having the 5V a bit leveled up by the USB kind of help PIco.

For all Rpi I use, for each, a [5V 3A power charger](https://www.amazon.fr/Aukru-Raspberry-Alimentation-Interrupteur-Longueur/dp/B01M58O9M9). I do think the issue is somehow related to power, but 5V 3amps is what you recommend, and I don't think there is 5V 4amps charger anywhere. Any idea how to solve my boot issue?

Thanks

PiModules, 2018-11-19:
Hi,

Please find my answer in the *text*

>
> Hello.
> I have a bunch of Rpi and PIco UPS to protect them, and some have 
> undervoltage issue upon boot:
*OK*
>
>   * Rpi3b + PIco UPS for mediacenter. So far, never had any issue
*OK*
>
>   * RPi3b+ + PIco + 2.13in epaper screen for PiHole: when booted up,
>     soon I get an "undervoltage log" and the rpi shutdowns after 40
>     seconds. To have a stable boot I need to:
>       o plug the power. The Rpi starts
>       o when I get the undervoltage error, unplug the 5V cable.
*Please explain in more details ***above sentence *to further support 
you. Please advise what firmware do you have as also what model*
>
>  *
>       o wait for a few seconds, plug again, boom, the status script
>         says "Rpi powered", and could stay like this for weeks
*Please explain in more details ***above sentence *to further support 
you. *
> *
>   * Rpi3b+ + PIco + USB keys (zwave sticks, rfxcom) for domoticz. Same
>     issue, undervoltage log and auto-shutdown after 40 sec. The
>     previous tricks does not work, I have to plug a [...]
*It could be the PSU, as many, many fake PSU are on the market. It could 
be also the micro USB cable, so PSU is fine but powering cable is weak.*

*Please provide me as much as possible information to support you, 
pictures of the PCU and its cable.*

*Kind Regards*

> Thanks
-- 
Ioannis
www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

gsemet, 2018-11-19:
Hi. I think the PSU is the culprit, indeed, do you have one recommendation for one that will work with the Rpi3b+, the PIco UPS (UPS PIco HV3.0B+ HAT Top-End 450) and another hat on top of the gpio connectors?

My orders was: 3341 (2x UPS PIco HV3.0B+ HAT Top-End 450 with Rpi3b+). I think I used the PSU provided by you (PSU \ no Power Supply Unit - EU Plug 5.1V 3A /1m cable). So, works on Rpi+PIco, but as soon I had the escreen (which does not use a lot of power), it show this undervoltage issue error somehow. And probably the firmware choose to shutdown the rpi instead of retesting the voltage ?
I do not have the firmware version here. I'll update this ticket tonight :)