modmypi/PiModules #39: fssd protocol documentation?

selli69, 2019-02-23:
Hi there!

This "issue" might be a little OT but because of the broken forum registration on the pimodules forum I try to get help here.

Is there any documentation, timing sheet, etc. for the fssd protocol on GPIO22/27? I found no information, neither in the manual, nor in the forum.

The python script you provide makes no sense to me, but I assume it's working as no one complains about.

Background for my question is, that I plan using my own daemon, which uses the pigpio library/daemon and a concurrent use with the RPi.GPIO library is not an option for me.

Thanks in advance!

cromarty, 2019-02-24:
This is from memory since it is a while since I wrote the daemon...

First of all, of course it works.  I tested it thoroughly before it was
released and there are scores of people using it.

There is no 'protocol' as such.

I seem to remember this is how it works:

Pin 27 is the clock pin.

Pin 22 is the pulse pin.

The pulse pin has on it a square wave created within the daemon process.
 If I remember correctly the upspico uses this signal to tell whether or
not the Pi is still running.  I think there is at least one LED on the
upspico board that flashes in sympathy with this square wave.  I can't
remember clearly, and can't test this again now quickly as I don't
currently have a upspico set up.

If power is lost the upspico uses the clock pin to signal a shutdown,
which is detected by the daemon, which runs the isr, interrupt service
routine.  If I remember rightly it is a falling-edge that triggers the
shut-down.

Please note that just because you don't understand the code does not
mean it is rubbish or that it does not work.

Good luck writing your own daemon.

Just for a note, the daemon is written for Python 2.7, which will reach
'end-of-life' in January 2020 and stop receiving updates or maintenance.

To change the code for Python 3:

1. Change all occurrences of "print 'somthing'" to "print('something')".
This will still work under Python 2.7, but will now work with Python 3.

I also recommend changing any occurrence of:

#!/usr/bin/python

to:

#!/usr/bin/env python

This will allow the code to run in a virtual environment.

Mike

If the power disappears, the upspico changes the state of pin 22 and

On 23/02/2019 23:20, selli69 wrote:
> Hi there!
>
> This "issue" might be a little OT but because of the broken forum registration on the pimodules forum I try to get help here.
>
> Is there any documentation, timing sheet, etc. for the fssd [...]

-- 
Michael A. Ray
Analyst/Programmer
Witley, Surrey, South-east UK

"Perfection is achieved, not when there is nothing more to add, but when
there is nothing left to take away." -- A. de Saint-Exupery

https://cromarty.github.io/
http://eyesfreelinux.ninja/
http://www.raspberryvi.org/

PiModules, 2019-02-24:
Hi,

Yes, we have a huge problem with our forum, please use this forum until 
we recover our own, or send direct emails to support@pimodues.com.

For the interaction with Raspberry Pi we are using the following below 
described protocol. It is interrupt based on both sides (PIco and RPi), 
and it has been selected to avoid any delays (like happens with former 
versions of PIco (not the HV3.0), where when coping of huge number of 
files could make PIco crazy (as the response of the "normal" script has 
been delayed). Please also kindly notice that we are working on the new 
updated Daemons, that will also have moved (as an option) the 
interrelation with raspberry Pi to I2C. That will release these 2 GPIOs 
for other applications.

The current (on 24.02.2019) version it is implemented by using 2 pins 
(GPIOs) the GPIO22 and GPIO27

It is interrupt driven on both sides (raspberry Pi and PIco).

The GPIO 27 (pin 14 on 40 pins connector) send pulses (pulse train) to 
the Raspberry Pi. They are firing an interrupt on the Raspberry Pi 
handled by Daemon.

This interrupt "response" with opposite level on the GPIO 22 (pin 16 on 
40 pins connector. The "response" is firing an interrupt on the PIco 
handler in the firmware. This interrelation guarantee the real time 
knowledge of the PIco that Raspberry Pi is running. In addition if PIco 
require to shutdown Raspberry Pi, hold the GPIO 22 low for a single 
cycle. Therefore before the Raspberry send to this GPIO 22 the opposite 
answer check before if it is made "low" by PIco, and if so, initiate the 
FSSD (file Save Shutdown process).

The Raspberry Pi respond always with 1/2 frequency of received pulses 
(obviously).

I will include in the new manual I'm writing (updating with all 
features) the graphical picture showing it.

Hope it helps

selli69, 2019-02-24:
Hello Mike!

Thanks for your quick reply and the effort to do this on a sunday!

> Please note that just because you don't understand the code does not mean it is rubbish or that it does not work.

I never said that.

What I don't understand is when only a falling edge on the clock pin is handeled by the isr routine, how the square wave is generated. The code for square wave generation can (as I understand how the callback works) never reached. But this may be a python/RPi.GPIO oddity.

My only question now is, what is an appropriate frequency for the square wave on GPIO22?

> Good luck writing your own daemon.

Don't worry! I wrote several daemons/services.

And thanks for mentioning some python specific stuff, but I do my programming in C++.

Cheers!
selli

PiModules, 2019-02-24:
_Follow Mikes, he is the author of the Daemons we are using !!_

selli69, 2019-02-24:
Hi Ioannis!

Thanks for your reply! I am astonished about your and Mikes efforts to do support on sundays!

All my questions are answered by your post. Now I understand Mike's code. :)

Look forward to a new customer!

Regards,
selli

cromarty, 2019-02-24:
Hello,

I'm struggling to remember exactly how the Python daemon works.

I used Python rather than C because that is what Io wanted, and because
email alerts were a lot easier to configure and run in Python than C.

Personally I prefer C or C++.

I'm also not currently involved with writing the new version.  I only
wrote the old version which Io has spoken about.

Most code for peripherals which are sold to makers is written in Python,
since the purpose of the Pi is primarily as a learning tool, and writing
it in Python makes it much easier for customers to modify the code, as
well as learn from it.

There is some commentary in the picofssd code, and more documentation in
the repository.

Mike

On 24/02/2019 18:01, selli69 wrote:
> Hello Mike!
>
> Thanks for your quick reply and the effort to do this on a sunday!
>
>
> I never said that.
>
> What I don't understand is when only a falling edge on the clock pin is handeled by the isr [...]

-- 
Michael A. Ray
Analyst/Programmer
Witley, Surrey, South-east UK

"Perfection is achieved, not when there is nothing more to add, but when
there is nothing left to take away." -- A. de Saint-Exupery

https://cromarty.github.io/
http://eyesfreelinux.ninja/
http://www.raspberryvi.org/

selli69, 2019-02-24:
Ah, one thing for Mike:

Take a look at pigpio. It's so much better than the RPi:GPIO library, it utilitizes I2C and has a python library. Maybe a good idea to use it for your upcoming new daemon for the new firmware versions Ioannis mentioned.

http://abyz.me.uk/rpi/pigpio/index.html

Cheers!

selli69, 2019-02-24:
@mike:

I know that many sees the Pi for learning. I use it with codesys as an PLC for non critical applications. With an UPS like the Pico it's an alterternative to much pricier PLCs.

closed 2019-02-24 by selli69

cromarty, 2019-02-24:
Thanks, I will look at pygpio.

It is a long time since I did any GPIO stuff.

On 24/02/2019 18:21, selli69 wrote:
> Ah, one thing for Mike:
> 
> Take a look at pigpio. It's so much better than the RPi:GPIO library, it utilitizes I2C and has a python library. Maybe a good idea to use it for your upcoming new daemon for the new firmware versions Ioannis mentioned.
> 
> http://abyz.me.uk/rpi/pigpio/index.html
> 
> Cheers!
> 

-- 
Michael A. Ray
Analyst/Programmer
Witley, Surrey, South-east UK

"Perfection is achieved, not when there is nothing more to add, but when
there is nothing left to take away." -- A. de Saint-Exupery

https://cromarty.github.io/
http://eyesfreelinux.ninja/
http://www.raspberryvi.org/

cromarty, 2019-02-24:
Hello,

I seem to remember that when I was writing the daemon there was a chap
who was using the UPS with a Pi in some industrial application where the
Pi was up a mountain somewhere doing something critical.

So yes, there are certainly uses for it.

I am currently creating a Docker container image with the current
picofssd in it.

Mike

On 24/02/2019 18:31, selli69 wrote:
> @mike:
> 
> I know that many sees the Pi for learning. I use it with codesys as an PLC for non critical applications. With an UPS like the Pico it's an alterternative to much pricier PLCs.
> 

-- 
Michael A. Ray
Analyst/Programmer
Witley, Surrey, South-east UK

"Perfection is achieved, not when there is nothing more to add, but when
there is nothing left to take away." -- A. de Saint-Exupery

https://cromarty.github.io/
http://eyesfreelinux.ninja/
http://www.raspberryvi.org/