modmypi/PiModules #40: Firmware update for HV3.0B

boydhako, 2019-04-10:
Are there any firmware updates for HV3.0B? I know there's updates for A.

PiModules, 2019-04-10:
Hi,

All the latest updates we put here

https://pimodules.com/products/ups-pico-hv3-0a-b-b/common-updates

That include manuals, firmware, ready SD cards, as also any other news

Each firmware is compatible with all models. Differences between models 
(A/B/B+) or (Plus/Advanced) are stored permanently in the bootloader. 
Therefore each firmware is "smart" and "know" on which hardware is running.

FYI

there are a new firmware covering practically all open issues 0x50, but 
not tested enough to release it to publicity, I hope that we will 
release it next week.

Kind Regards

Ioannis

daghemo, 2019-05-14:
How can we access the firmware only and the user guide?

Ve2mrx, 2020-04-28:
Any new firmware update? 0xE3 is nowhere to be found, and I cannot find documentation.

I'm still waiting for the serial port fix and Raspbian Buster support. I have a UPS PIco HV3.0B+ HAT Top-End 450.

Thanks, Martin

daghemo, 2020-04-29:
Hi @Ve2mrx,
I think that you can find the latest firmware within the ZIP linked at the top of https://pimodules.com/common-updates (released in March of 2019).
I had some problems with my two UPS PIco HV3.0B+ HAT Advanced 450, so I've switched to other UPS solutions for my Raspberry Pis and I'm now very happy!

Ve2mrx, 2020-04-30:
Hi @daghemo,
Well, that's a NOOBS SD card image, 4GB+. And iirc, the 0xE3 firmware was from 2019-07. 

In a way, what I was saying is that the links are dead, the forum is gone, and if I want to make this board usable, I *MUST* have a proper firmware that effectively disables the serial port on the UPS hat so I can use /dev/ttyAMA0 with my GNSS receiver. Because of this, I never used the Pico.

Last I read about 0xE3, it was in beta. Now, there is no replacement released after close to a year. Luckily, I have saved a copy of 0xE3 a while ago, but I'm trying to see if the hardware should be considered abandoned or not. Next step is email, not hoping for much at this point.

Martin

daghemo, 2020-04-30:
Hi @Ve2mrx,
I'm not "affliated" with Pi Modules and I perfectly understand your position.
I had the same problems in the 2018: I bought some boards, then asked for a refund. :( 
You may found other issues from me here. :(
I still have two of them, just because they were installed in a building site before full testing and I have no access to them anymore.  In the last two years I found other UPS solutions for my Raspberry Pi. Me too, I need both GNSS and 4G/LTE. :)