modmypi/PiModules #43: Unable to access the serial port when Pico HAT connected

mikcatta, 2019-05-07:
I can read my serial GPS fine, without the Pico HAT connected. However, as soon as it is installed, access to the GPS on ttyAMA0 results in nadda.

The documentation suggests that the serial port is only enabled during boot and f/w upgrade. 

I have tried "sudo i2cset -y 1 0x6b 0x02 0x00" to disable the serial port on the Pico, without any joy.

PiModules, 2019-05-08:
Hi,

This bug has been reported to us and already corrected. I will release 
beta version with this patch today or tomorrow

This command is disabling the Serial Port, however due to bug in 
compiler in some cases, a low level TX was present (about 300 mV) this 
cause the problem.

Kind Regards

mikcatta, 2019-05-08:
f/w  0xE3 seems to of resolved this issue. https://forum.modmypi.com/viewtopic.php?f=33&t=9770&p=18131#p18131

closed 2019-05-08 by mikcatta

PiModules, 2019-05-08:
the proper is done now, as I said, in some cases is OK, in other NOT, 
due to bug in compiler. We done the patch, and release today or tomorrow 
in our website www.pimodules.com (in are when updates are released)

mikcatta, 2019-05-09:
I am not seeing anything new on this page https://pimodules.com/news/firmware-updates?

PiModules, 2019-05-09:
Hi,

Was overloaded with new production, release it during the weekend. Just 
testing n various environment to be sure

Kind Regards

-- 

Ioannis
www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

Ve2mrx, 2020-04-15:
Any update on this firmware? It's not on the website's firmware or common sections.

mikcatta, 2020-04-16:
I gave up on this product, it seemed support was poor.

Ve2mrx, 2020-04-16:
Well, mine has been sitting next to the Pi for close to a year as I need the serial port for GNSS, and it's hogged by the board. I bought a new Pi 4 and a UPS hat from BuyaPi.ca, much simpler with no serial port hogging. 

It's somewhat easy to hook together parts to create a great product on paper, but here, the complex part is the firmware and software, which don't work right. If only we had firmware source, we could help! Just slap the right license on it :-)

I want to give one last chance before I change my original plans.

Martin Boissonneault
Sent from my phone