modmypi/PiModules #45: Raspberry doesn't start with power cable

mammnon, 2019-06-14:
Hi,

I installed a brand new pico hat on a brand new raspbien. I did all the installation steps, installed the "on/off" button and the buzzer.
First time I plugged the power with the hat, the raspberry started and the pico worked perfectly (I removed the cable and the raspberry continues to work).
Then I shutdown the raspberry (doesn't matter if I do it pushing apico button or in command line).
I tried all buttons the raspberry never started again. I have to unplug the power cable only then a button started the raspberry.

All I want is: If the power cable goes off=> shutdown the rasbperry (no wait)
If the power cable goes on=> start the rapberry.

I don't want to have to use buttons for start/stop, any chance to setup this behaviour? Or at least to be able to start the raspberry with the power cable on?

Thanks in advance.

PiModules, 2019-06-14:
Hi,

Hmm, can you please send me a short video to see this effect.

Also, can you please make factory reset as described in the manual.

I need more description of your problem in order to support you 
properly. The best is in bullets, step by step what happen (what you done)

The short video would be also the best - 20 seconds. Also LEDs are doing ?

Do not worry, if needed we will send you replacement.

Please me your details to support@pimodules.com

BR Io

mammnon, 2019-06-15:
More details as requested:

- rapsberry is ON (doesn't matter how).
- Power cable is ON (same while OFF btw)
- sudo halt 
- wait until complete shudown
- power cable is removed
- power cable is plugged-IN
       => nothing happen (I would expect the raspberry to start...)
- I use the 2 user PICO buttons
      => nothing happen (I would expect the raspberry to start ...)
- power cable is removed
- I use the 2 user PICO buttons
    => The raspberry start. 

So in conclusion => the only way to start the raspberry is to have no power supply. Once started I can plug in the cable? Seems a bit weird, isn't it?

The video will be added, but there is no added value.
I push the reset button on the pico (the third one), is it enough to make the hard reset ? It maked the pi shutdown btw...

mammnon, 2019-06-15:
video : https://drive.google.com/file/d/1-UXiserfNH4xQZWImTtBm5fuosgIfc3f/view?usp=sharing