modmypi/PiModules #48: Battery version 0x43??  And swelling 450mAh LiPO battery.

mminehanNZ, 2019-09-05:
I am new to UPS PIco world.  I just received 2 boards from PiModules (UPS PIco HV3.0B+ HAT Stack 450 PoE).  Took a while to get them up and running.  I've been making up circuits since my teenage years and am not an amature at soldering components.  The 'Magic Switch' was a bit confusing as the manual has few errors and is missing some information.  Strange rebooting, not staying shut down and boot loops made it all very frustrating.
The manual really needs checking and updating.   E.g. from page 83:

> When System is Cable Power user has a possibility to restore the factory defaults. To do that the following steps need to be followed:
> • Press and hold the UR button
> • Continue to hold the UR button, and press and hold the C button.
> • Release the UR button, but keep holding the F button
> • Release the F button

What?  This does not make sense.  Keep holding the F button, or should that be C button.

Took hours to figure things out.  Not the most fun thing I have done in a while.

Running the pico_status_hv3.0.py I get the following errors:

> UPS PIco Firmware.....: ec
>   UPS PIco Bootloader...: 53
>   UPS PIco PCB Version..: 43
>   UPS PIco BAT Version..: ERROR
>   UPS PIco BAT Runtime..: 2 MIN
>   UPS PIco r232 State...: OFF
> 
>   Powering Mode.........: RPi POWERED
>   BAT Percentage........:
> Traceback (most recent call last):
>   File "pico_status_hv3.0.py", line 194, in <module>
>     print " ","BAT Percentage........:",bat_percentage(),"%"
>   File "pico_status_hv3.0.py", line 95, in bat_percentage
>     return datapercentage
> UnboundLocalError: local variable 'datapercentage' referenced before assignment

So I had look at the code and tried to work out what is going wrong.  Turns out I have a battery type 0x43:

> sudo i2cget -y 1 0x6b 0x07
> 0x43

I tried resetting the board to the factory defaults using the buttons as per the manual but still get the same battery code.

Is this a bug?  Have the battery codes changed in version ec?
According to the manual this is an invalid battery type.
Should I try and change the code to 0x53?

Also.......one of the batteries is swelling/ballooning like it's filling with gas.  Is this normal.  The voltage reads (sudo i2cget -y 1 0x69 0x08 w) as 0x0440 which I presume in 4.4V.  Max voltage should be 4.2v for a LiPO.

And when will the official forums be fixed, and the new firmware (0xed) released which has been in beta for almost a year!  Using GitHub as forum is not ideal.

So far I would give the whole UPS PIco experience a 5 out of 10 rating.   I'm kind of wishing I had tried a different UPS HAT.  But it's hard to find one that works with the official Pi PoE HAT.

mminehanNZ, 2019-09-06:
So I found a better fork by janmagnet.  And I change the battery type to 0x53 as per the manual.
Still had no response to my email to PiModules about the battery type register and swelling battery. :(

mminehanNZ, 2019-09-09:
I have added a sub fork to the janmagnet fork.  This includes a pico_status script that reads back the system information register.

tetzlav, 2019-09-11:
I have the same frustrating experiences with UPS Pico (Stack PoE Version). Battery type 0x43 etc...

But my first problem is, that the Pico doesn't recognize RPI powering. and the BAT level is always 0x000 and RPI powering level 0x500 (=5.00V). I measured the voltage levels with external equipment. BAT level is 4.04V, RPI powering level is 5.05V.

Do you have made some other register settings or is my Pico scrap?

Overall i think the Pico project is dead!
Ioannis Mourtsiadis muddling along a bit, sells the rest of Picos he have and there is no real support anymore. To silence the frustrated users the forum was shut down and emails are not answered. It wasn't better with the old Pico versions 1.0 an 1.1. The Picos never really worked like charm.

PiModules, 2019-09-11:
> Ioannis muddling along a bit, sells the rest of Picos he have and

Hi,

First of all as I mentioned that we are supporting PIco over support 
email, as it can be monitored, forums re good tool, but difficult for us 
to be monitored.

Therefore support done over the support@pimodules.com is the best tool.

Back to your technical problem, battery type is set to LiPO, and as far 
you are using LiPO battery it is the same. The different type was setup 
in order to calculate running time based on capacity. Therefore use 
battery as it is, except if you are selecting other chemistry type. It 
is also good to declare the firmware version, so we can easier to 
support you.

What is the difference between PoE PIco and stack. The difference is 
different resistor that decrease the battery charging current, to 
smaller one.

Why we are doing that, because power supplying capabilities are 
restricted in the PoE circuit. BTW. Please also report if you have the 
old PoE or the newer one improved.

Please report or send video, how your UPS, BAT and CHG led behavior. 
Description is usually not enough. You have 5V because your system is 
supplied by PIco, and NOT PoE. The most possible reason, without knowing 
what you done is that your daemons are not running properly.

Therefore in order to clarify it, please send me or to forum a short 
video (10-15 seconds)

You do not need to set anything for normal PoE setup, but I need to know 
if you have normal setup. There are o special registers.

PIco Project is not dead !! Definitely not !!, We are releasing new PIco 
HV4.0 fro the new Raspberry Pi, new firmware, but we are small, small 
company so many times have delays.

We have a problem with manufacturing of good qty PIco in order to cover 
requests than to sell rest of them as there are no rest of PIco, as you 
can see only few units there are. Therefore please not do such 
assumption, especially in publicity.

Please answer to my questions, and then I will drive you to solve your 
problem.

Kindly Awaiting for your detailed information instead if general one in 
order to further support you

Warmest Regards

Ioannis

PiModules, 2019-09-11:
Hi,

The only reason for swelling battery is high temperature or long live.

Please advise your configuration and order time

Kind Regards

Ioannis

PiModules, 2019-09-11:
Hi,

Thank you for your post.

Please fin my answer within *your text*

>
> I am new to UPS PIco world. I just received 2 boards from PiModules 
> (UPS PIco HV3.0B+ HAT Stack 450 PoE). Took a while to get them up and 
> running. I've been making up circuits since my teenage years and am 
> not an amature at soldering components.
*You can always ask to solder these few components if you cannot do it 
your self. We are offering such service in a very low price*
>
> The 'Magic Switch' was a bit confusing as the manual has few errors 
> and is missing some information.
*I assume that your PIco is working now properly, and you are trying 
other functionalities. Please clarify ?*

> Strange rebooting, not staying shut down and boot loops made it all 
> very frustrating.
*Have you soldered the POGO pins ?*
> The manual really needs checking and updating. E.g. from page 83:
>
>     When System is Cable Power user has a possibility to restore the
>     factory defaults. To do that the following steps need to [...]
*OK. our mistake, should be C,  you are right*

*Thee are bout 200 pages, so can be some mistakes*

> Took hours to figure things out. Not the most fun thing I have done in 
> a while.
*Please do not use this script. One customer released it, but it is full 
of bugs , we are going to release our own with this ongoing firmware. 
This script was working for former version of PIco, *
>
> Running the pico_status_hv3.0.py I get the following errors:
>
>     UPS PIco Firmware.....: ec
>     UPS PIco Bootloader...: 53
>     UPS PIco PCB Version..: 43
>     UPS PIco BAT Version..: ERROR
>     UPS PIco BAT Runtime..: 2 MIN
>     UPS PIco r232 State...: OFF
>
>     Powering Mode.........: RPi POWERED
>     BAT Percentage........:
>     Traceback (most recent call last):
>     File "pico_status_hv3.0.py", line 194, in
>     print " ","BAT Percentage........:",bat_percentage(),"%"
>     File "pico_status_hv3.0.py", line 95, in bat_percentage
>     return datapercentage
>     UnboundLocalError: local variable 'datapercentage' referenced
>     before assignment
>
> So I had look at the code and tried to work out what is going wrong. 
> Turns out I have a battery type 0x43:
>
>     sudo i2cget -y 1 0x6b 0x07
>     0x43
>
> I tried resetting the board to the factory defaults using the buttons 
> as per the manual but still get the same battery code.
>
> Is this a bug? Have the battery codes changed in version ec?
> According to the manual this is an invalid battery type.
> Should I try and change the code to 0x53?
>
> Also.......one of the batteries is swelling/ballooning like it's 
> filling with gas. Is this normal. The voltage reads (sudo i2cget -y 1 
> 0x69 0x08 w) as 0x0440 which I presume in 4.4V. Max voltage should be 
> 4.2v for a LiPO.
*The battery resume A/D is based on resistor divider, and as an internal 
reference has micro controller reference. Therefore it is possible to 
have some deviations on that. It is not a multi-meter. The charging is 
done by different IC, that is measuring eh battery level internally and 
very accurate*

*We are going to add system self adjusting to have more accurate measures.
*

> And when will the official forums be fixed, and the new firmware
> (0xed) released which has been in beta for almost a year! Using GitHub
> as forum is not ideal.
>
> So far I would give the whole UPS [...]
*Have you installed both daemons ?*

*If your PoE is giving lower voltage you may need to adjust the level of 
firing of battery back up PAGE 84
*

*Kind Regards*

*Ioannis*

PiModules, 2019-09-11:
Hi,

Such type batteries are swelling due to 2 reasons.

1. Battery is old (more than 2 years), as we are ordering new batteries 
overtime for the production, I assume your battery is fresh

2. Battery has been overheated, this is possible, please advise when you 
placed the battery

I any case I will send you replacement, please contact with me at 
support@pimodules.com

Kind Regards

Ioannis

On 07-Sep-19 12:57 AM, mminehanNZ wrote:
> So I found a better fork by janmagnet. And I change the battery type 
> to 0x53 as per the manual.
> Still had no response to my email to PiModules about the battery type 
> register and swelling battery. :(

-- 
Ioannis
www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

mminehanNZ, 2019-09-12:
> I have the same frustrating experiences with UPS Pico (Stack PoE Version). Battery type 0x43 etc...
>
> But my first problem is, that the Pico doesn't recognize RPI powering. and the BAT level is [...]

Hi @tetzlav 

I have had more success recently.  I got a response from Ioannis.  I think a big part of the problem is that there are a few errors in the manual which needs updating.

0x43 apparently is a valid battery format.  But this is not mentioned in the manual.  And the GitHub modmypi pico_status script does not recognize this format.  I have a separate fork which I will update in the next day or so to allow 0x43.

I did change the battery type to 0x53 by using _sudo i2cset -y 1 0x6b 0x07 0x53_

What register are you getting for the battery voltage? What do you get if you type _i2cget -y 1 0x69 0x08 w_

What type of battery are you using?

I agree that forums are a much better way to support a product, especially these kind of products.  End users can support each other and often solve problems.

Another issue is the size of the battery.  I found that once left the batteries to fully charge they seem more stable.  I think the 450mAh battery is too small.  It takes a bit of time after a power failure (FSSD) for the battery to fully charge if you have the PoE Pico.

The UPS Pico is a powerful board.  I can't find any other Raspberry UPS that supports the genuine Raspberry PoE HAT.  And very few of the other Raspberry UPSs support restarting without having to push some sort of button.  The UPS Pico makes use of both PEN and RUN pogo pins which allows for 'soft restarts'.

@PiModules 

Thanks for offering to send a new battery.  I have already found a replacement.  The faulty battery was only 2 days old and was not getting hot.  It was set up in a open environment, not in a case.

I have no problem with soldering myself.  The pogo pins were a little bit tricky, but they are seated properly and working.  I tested the conductivity withe a meter.

Yes...both daemons are running:

> pi@raspberrypi2:~ $ service picofssd status
> ● picofssd.service - LSB: PiModules(R) UPS PIco file-safe shutdown daemon
>    Loaded: loaded (/etc/init.d/picofssd; generated)
>    Active: active (running) since Thu 2019-09-12 02:22:43 NZST; 11h ago

I have offered to help Ioannis with the manuals.  Hopefully this will help other users.

tetzlav, 2019-09-13:
> [...]
> 19. UPS PIco manual is 200 pages and still growing, yes there are some small mistakes, but OK, it is 200 pages

Yes, i think the functions and registers in hardware and firmware grown too complex and the general documentation is not up-to-date. You can not rely on this informations (because of old informations and mistakes). Even the official tools here at github are not-up-to-date. They dont even know the default battery type and show ERRORs in default setup. Thats not really professional...

> 20. UPS PIco is not a toy, and it is addressed to professional users, especially due to many additional functionalities that has been implemented inside. 

And yes, in times of Pico HW 1.1 i tried to implement the Pico in professional industrial envroment (as USV for a Raspi driven Panel that shows some process parameter and graphs to user/visitors) and i give up because of buggy firmware. For examble the battery was killed again and again (and swelling) because of not reliable powering, battery and charging state detection...

> 21. The UPS PIco is not dead, as mentioned one of anonymous user here on this forum, and using my non anonymous personal data, it is continue developing and new firmware release is ready for about 2 months to be released, just under very extensive testing

The "new" firmware is announced since ONE year in beta status. I wonder if she'll come (within next days).... It was the same as in HW version 1.0/1.1/1.2 times. 

Exemplary you can read the thread [New Firmware 3.0 for the UPS PIco HV1.0 and HV1.1](http://108.175.10.239/viewtopic.php?f=12&t=7138) in (meanwhile readonly) forum, or this threat [Update of firmware to support latest kernel ??](http://108.175.10.239/viewtopic.php?f=12&t=6233) and in general all the problem posts in [the old UPS Pico subforum](http://108.175.10.239/viewforum.php?f=12). Some threads even appear to be deleted - why?

I also miss a firmware archive with traceable changelogs of firmware. I remember that e.g firmware 0x53 worked fine with hw 1.1 - 0x59 no longer. Infos are bit here at github wiki and on the hompage and in old forum threads and on google docs and as issue at github - its very confusing!

> 22. Yes, We are going to release new UPS PIco HV4.0, there ware many problems with temperature of Raspberry Pi4, battery thermal isolation, full 3A battery back up current, and at the same time keeping the price same low as before. We solved all that problems on the new UP PIco HV4.0 and added much more functionalities, keep the compatibility with old models, it is under production and very soon (hope next week) we will start pre-orders. 
>
>23. Before we will start pre-orders of the new PIco we will release the new firmware update for the old one
>[...]

Maybe you should begin to cleanup your official Homepage (dead links, actual infos, ...) first, maintain your tools and wiki at github and fix your firmware instead to use your energy develop new functions and hardware releases.

I own different hardware releases of UPS Pico. No one worked out-of-the-box. Everytime i had hoped know every is fine, new hardware, new firmware, new software tools - but years later everthing is the same...

As i already said: the Pico CAN be nice piece of hardware. In there is a bit brain work and much know-how. But without software, firmware and (community) support its useless, And this part ist unfortunately not at hardware level.

^^^^^^^^^
MY OPINION!

mminehanNZ, 2019-09-25:
@PiModules   I did NOT say "I have the same frustrating experiences with UPS Pico (Stack PoE
     Version). Battery type 0x43 etc..."
You have quoted the wrong person.  Look at the 4th post and you will see that is not me.  Some how you have it wrong.

When is the new firmware coming out?  I am still having issues and am hoping the new firmware will fix some of these.  One module reboots because of a 'MCLR' issue that I am still trying to figure out.  The other now has a restart_watchdog problem after rebooting sometimes.

You really should set up a proper forum.  GitHub is not the place for users to look for hardware support.  And email support can be very slow and frustrating.  Forums work and they allow community support.

PiModules, 2019-09-25:
Hi,

We have it ready, a long time, just under very extensive testing as we 
rewritten some critical routines. We expected to release it this 
weekend, but I stopped it as need more analysis. The current release 
date is this weekend, and I do not think that it will be changed 
anymore. This ongoing version is the 0x50, where we cover all pen issues.

Regarding the forum, we recover it but not moved yet to the our domain, 
hope and wish to do it very soon

Temporary please use support@pimodules.com

Regards

tetzlav, 2019-10-08:
> Hi, We have it ready, a long time, just under very extensive testing as we rewritten some critical routines. We expected to release it this weekend, but I stopped it as need more analysis. The current release date is this weekend, and I do not think that it will be changed anymore.

We are waiting patiently for the new release...

PiModules, 2019-10-09:
Hi,

Tomorrow max after tomorrow.

As some users have had problem with adjusting of levels (seems you too, 
sorry about that) I added a self adjusting system, so you not need to do 
anything anymore. System will self adjust to existing voltage levels of 
PoE, weak PSU etc

BR

mminehanNZ, 2019-10-13:
> Tomorrow max after tomorrow.

And....still nothing.  Seems to a recurring theme.

I'm going to start looking at other options for a UPS for the Pis.  I really just need a reliable UPS that can safe shutdown.  All the other stuff like sounder, LEDs, fan etc are nice but not essential.  What is essential is a UPS that works.

I'm still having issues with random shutdowns reported as MCLR.  And sometimes when I manually shutdown the PIco automatically restarts the Pi.  This is happening on two different setups.

I was hoping the new firmware was going to fix some of these issues but I'm tired of waiting.  Time to move on and find a different UPS. 

Good luck everyone.

tetzlav, 2019-10-19:
> Hi, Tomorrow max after tomorrow. 

It seems to be a joke!
Im stop waiting too and the PiCo will be scrapped... :-/

mminehanNZ, 2019-10-21:
I just got hold of PiJuice UPS,  It seems way better than the PIco module.
The software is much better.  Seems more stable.
I had to uninstall the PIco daemons, change the /etc/modules (leaving the i2c-dev) and update-rc.d etc.  Basically work through the installation in reverse.
I'll let you know how it goes.
And best of all I can confirm that the PiJuice works with the official Raspberry Pi PoE HAT.

PiModules, 2019-10-21:
Hi,

We have glued on the automatic adjustment procedure that has been added 
especially for you and some other people that can not (as it looks like) 
to adjust their system to the existing environment)

By construction it is not possible to have reported MCLR, as the only 
way to have MCLS is to press UR button. You must be doing something 
wrong. As far you are using a weak power source after a time system will 
be not able to support you, it is logical. You need to have do a proper 
setup for your specific Power monitoring i.e.

sudo i2cset -y 1 0x6b 0x1a 0x25
sudo i2cset -y 1 0x6b 0x1b 0x03

or even increase the second to

sudo i2cset -y 1 0x6b 0x1b 0x04

This is the penalty if using the PoE as only powering system and not 
other additional PSU. You need to adjust little bit to your existing 
powering conditions. Please also confirm that you not have an old 
version of PoE, as there ware some problems with power loads.

Your system is definitely not changing the battery, and on my opinion, 
it is not charging because power is no sufficient, so you are using your 
system with discharged battery. You need to make proposed changes.

If it is possible please send me a video of your LEDs to see how they 
are blinking, this will allows me to estimate more accurate your problem.

BR

PiModules, 2019-10-21:
Hi,

Under normal conditions, it is forum for technical support of UPS PIco, 
however as you used this forum for other advertising I would need to 
answer to this also. I'm not willing to put one product against the 
others however you are not providing accurate information on this.

There are many UPS for Raspberry Pi on the market. However all of them 
are using separate power input and not monitoring the GPIO by itself.

This kind of UPS offered by others is called - off line UPS - as are 
using separate power input and when this power input is absent they 
switches to battery back up. They use the separate powering input 
(instead of Raspberry Pi micro USB one) and are powering the system via 
GPIO 5V. Therefore absence of power available on the input firing the 
battery back up. You need on all other UPS for Raspberry Pi (as you see 
there are always micro USB input on PCB) to have a dedicated case. These 
types of UPS can not work with PoE.

The UPS PIco HV3.0 is more closed to online UPS or line interactive UPS 
as it s monitoring the powering line on GPIO 5V itself (there is no need 
to addition power source - additional micro USB on PCB, so can be used 
with any case available on the market) and power gaps on the GPIO 
automatically switches to the battery back up. This is described in very 
details in the manual.

Mentioned by you by construction (as far I know) is using separate input 
for powering input. Therefore you will need to have connected in 
additional to PoE also external PSU, as it is not based on monitoring of 
GPIO 5V. This is the main difference between PIco and all other 
Raspberry Pi UPS

The UPS PIco is using the power provided by PoE itself to powering the 
system. If you like to add external PSU you can do it also in PIco, but 
this is nonsense to use additional PSU on PoE system.

 From the beginning I asked you to adjust some of monitoring parameters 
of power monitoring, as far I understand you not done it. We sold couple 
of UPS PIco for PoE and no one reported problem, may be a few users.

As UPS PIco is monitoring the 5V GPIO it is mandatory to make some 
adjustment to your configuration (current consumption, and PoE voltage 
level)

As I seen, some (i.e. you) people can not (or not like) to do this 
adjustment, we are implementing an automatic adjustment routine.

BR

mminehanNZ, 2019-10-21:
Seriously?  This is your response to the problem?
You don't think the lack of a support forum (previously shutdown and never reinstated), blaming users, a manual with errors in it, firmware that has been repeatedly promised and never delivered, a website with numerous dead links, and some users reporting swollen batteries could be the real problem?

Blaming the end user:
> the only way to have MCLS is to press UR button. You must be doing something wrong.
No I am not touching any of the buttons.  It sits untouched in a server rack.  The PIco runs ok for 2 or 3 days and then the MCLR error appears in the SysInfo register.  And yes I know about the Rev2 PoE HAT.  I have definitely got the new versions.

Empty promises of firmware updates:
> Hi, Tomorrow max after tomorrow.
> The current releasedate is this weekend
Your website says 0xed 'still in beta' since August 2018!

No user forum:
> it is forum for technical support of UPS PIco
Which forum? Where?  The link on the Pi Modules website goes nowhere!

> This is the penalty if using the PoE as only powering system and not other additional PSU. 
If the PIco is not setup for a PoE supply by default then why make a big deal out of PoE support.  If you are going to sell a device with the main feature being PoE support then set it up that way.  It's too complicated to expect end users to have to experiment with pwr_spk_duration and pwr_rfc_level registers.

> This is described in very details in the manual.
The manual also says, in big red writing,  "These parameters should be only changed very experienced user, that are understand properly the Power Monitoring Algorithm, and are able to define properly the new conditions. For all other users, it is recommended to keep these parameters as they set by manufacturer."  Again, not very easy for end users to understand what to do?  But hey......just blame the users.

> There are many UPS for Raspberry Pi on the market. However all of them are using separate power input and not monitoring the GPIO by itself.
This is not true.  The PiJuice uses the GPIO.  I have the PiJuice UPS setup with the official PoE HAT (no external USB power) and it is working perfectly.  It reads the supply voltage, battery voltage, battery temperature etc.  No need for a special case either.  You do not need to use the POGO pin with a 3b+.  And the software is easy to understand and use.  No need to manually set complicated registers.

> no one reported problem, may be a few users.
So in fact other people are also having problems.

I have spent enough time and money on the PIco.  You need to learn some lessons about customer service.  An apology for the delayed firmware would be a good start.  You need to fix your website, reinstate the user forums, fix the manuals, and release the so called 'updated' firmware, before you start making accusations like 'you must be doing something wrong'.   You are not going to have a successful business this way.

This whole experience has been very disappointing.  My advice to other users or people looking for Pi UPS is do your homework, read about issues and make sure there is adequate support (i.e. user forums and reliable firmware updates) for the device you buy.   The PIco has many great features, on paper at least.  But it is too complicated, difficult and time consuming for the average Pi user.

Photo below is of the PIco vs PiJuice.  Notice neither is using any USB power.  They are both being powered with official Raspberry Pi PoE HATS (updated versions).
[image]

PiModules, 2019-10-21:
Hi,

I'm not willing of course blaming you, just trying to solve your 
technical problem.

Only this. As your email is long, please find my answer in the text with 
*bold*

> Seriously? This is your response to the problem?
> You don't think the lack of a support forum (previously shutdown and
> never reinstated), blaming users, a manual with errors in it, firmware
> that [...]
*The only reason that batteries can be slowed is high temperature, or 
long time of usage, or very unusual bad battery. Anyhow I offered you 
immediately by email to replace the battery without relay looking for 
the reason. On personal email, you said that it is fine and not needed. 
As we have a high demand of PIco, batteries usually are here, no longer 
than 2 weeks. Therefore remain 2 other reasons high temperature or 
faulty batteries. Both are possible, as I m assuming you have not used 
them on high temperature, remain that battery was faulty. Very unusual, 
but always possible. That was the reason I offered you to send you 
replacement.*

*Regarding dead-links, we are a small company, mainly related to design 
and production (of course not only UPS PIco), so as we upgraded (by 
our-self the new website), some links are dead. We are working on that 
to update. Whatever customer are questing by email we are sending it.
*

> Blaming the end user:
>
>     the only way to have MCLS is to press UR button. You must be doing
>     something wrong.
*This is my opinion as I'm trying to see what is the problem on your 
side. From construction of PIco (the MCLR is connected to UR) therefore 
the most possible reason is that it has been used. We implemented a 
mechanism (you just used it) to check the reason of reset and other 
system information). I'm trying to indicate what could be the reason, 
not attacking you, just trying to see what could be wrong. The only 
alternatively could completely discharged battery, so PIco is switching 
to battery and as battery is empty, system is resetting. But then you 
should have beeping and BAT LED lit. You have not reported it to me.
*

*If you can please send me a short video of your PIco LEDs
*

>     No I am not touching any of the buttons. It sits untouched in a
>     server rack.
*OK accepted, then we should see what other could be wrong*
>
>     The PIco runs ok for 2 or 3 days and then the MCLR error appears
>     in the SysInfo register.
*So, as I mentioned you used the controlling mechanism. Do you have any 
other flags ON, on this variable (register) ?*
>
>     And yes I know about the Rev2 PoE HAT. I have definitely got the
>     new versions.
*OK. That is also good to know, but reported these info right now, I 
never received any information regarding these before on support email 
or forum. On PoE PIco we are using low current for battery charging just 
80 mA, as PoE is reduced with current availability*
>
> Empty promises of firmware updates:
>
>     Hi, Tomorrow max after tomorrow.
>     The current releasedate is this weekend
>     Your website says 0xed 'still in beta' since August 2018!
*OK. You are right, but please understand that the UPS PIco firmware is 
multitasking firmware with 10 Interrupts used, so debugging sometimes 
[... 50 lines cut ...]
> users or people looking for Pi UPS is do your homework, read about
> issues and make sure there is adequate support (i.e. user [...]
*You put me on a very difficult state to write against other company and 
product. I'm not working on that way. Of course I have not studied their 
product, however it is difficult to me to understand how it is possible 
to recover system when start shutting down and cable power returns 
without resetting the Raspberry Pi with gold pin and at the same time 
power the system via Raspberry Pi micro USB. It is just not possible on 
my opinion. But OK I probably missing something**
*

*OK. I got your remarks. **
*

*Kind Regard*s

> 20191022_093538 
> <https://user-images.githubusercontent.com/48574178/67242169-6dcc3900-f4b1-11e9-9ff3-b51e24432747.jpg>
-- 
Ioannis
www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

tetzlav, 2019-11-15:
> We expected to release it this weekend, but I stopped it as need more analysis. The current release date is this weekend, and I do not think that it will be changed anymore. This ongoing version is the 0x50, where we cover all pen issues.

...another month later maybe some less people are waiting anymore...

davros1973, 2019-11-15:
Hi

I came here with an issue and following the 0x43 lead via Google - but - turns out, **the issue was my confusion/mistake**.  Still getting to grips with the product (1.5 days in, so far). I've read through a few versions of the manual now, as well as threads in forums, various web sites and so on.  I'm even maintaining a spreadsheet now as reference material.

Reading this thread though ... (Ioannis) (this is just my personal experience/interpretation/opinion)  ...
 
I think the UPS PIco hardware specs look great.  I appreciate the hard work you've put into this product.

Why I decided on your product:

As an end consumer looking for a quick LiFePO4 reliable UPS to harden & use and to manage much of the power in my entire house including servers, with zigbee, hard wired connected devices (other bistable latching relays etc. and other logic-driven switching), alternative 868MHz connected, taking into account various sensor feedback, I couldn't see anything that could compete with using your hardware as a platform I could extend to all my use cases.  I want to make a personal app eventually to easily understand the state of all my equipment and servers etc. and if necessary automatically deal with brownouts/blackouts in a reasonable way, automatically but with remote manual intervention - but also reasonably hardened against malicious attack.  E.g. I always use an RC circuit on any higher-powered bistable latching relay so even if a malicious user obtained remote control of it, they wouldn't be able to switch it more than however many seconds or even minutes I deem reasonable for the device/load (but also thinking about arcing of course).  That's the sort of control freak I am.  :)

I worry about all the LiPO batteries I have in the house - using power banks with pass through inappropriately as well as so many other devices with them in held often at 100% charge.  I worry how they might present more risk over time.  So I wanted a ready made LiFePO4 UPS solution, that I could more or less forget about for 10-years say, and - that I could program fairly freely and use with other ready made (programmable) solutions so that I could minimise the work required to achieve my goals on a very money/time constrained budget.

Frustrations ...

Alas - I share some of the frustration aired here.  I wish you'd somehow find a very competent person who could revise all previous documentation related to the UPS PIco to achieve greater coherency and consistent accessibility for us (speaking for myself) amateurs.  That kind of undertaking is an art-form that not many companies achieve unfortunately.  It's not one of my skills.  I think there might be a slight language barrier sometimes when it comes to customer communication too, and maybe, that chasm in empathy that sometimes arises between a stressed/hard-working/proud tech guy/team and their ungrateful customers who don't seem to appreciate the effort put into the products and don't seem to listen to what the product owner thinks are the simple solutions.  (My interpretation of the different pov's).

I'm looking forward to the Raspberry Pi 4 implementation (whenever it comes).  (The Pi3 is actually a better choice for my application I think, anyway - but anticipate other uses for the Pi4 variant).

I appreciate you can't keep up with production as is ... I think that with better consumer/pro-sumer level documentation and supporting software and support for creating a market for 3rd party software and with better integration to other popular software then you could see even greater demand that might be worth scaling up production for reaching better efficiencies!  I think you could see better margins with more focus on software actually.  I'm saying this because I want to consume such offerings and am always frustrated when I can't find what I want on the market.  It's frustrating to see such great products that could be superb in every detail if only the will/vision was there.

So ... I'm a frustrated but still happy customer.  So far.  :)  I've had to sacrifice a lot of personal time on top of the substantial money I've already paid for the hardware  which is satisfying in some ways but not a priority among all my other time-constrained priorities ... (plus finding suitable reputable LiFePO4 cells in stock anywhere was a pain - I ordered from Italy on this occasion, and, had to order the HAT itself from Germany, and the other parts from my home - UK) ...  I would have preferred to have just paid a reasonable amount of extra money to get instant gratification with easy-to-use self-documenting software and easy integration with other products.  

I feel like it's been a very uphill journey.  I knew I'd have to do some soldering, but I didn't quite realise I'd have to parse so much documentation just to analyse current relevancy, or, to try to understand how 0x50 is greater than the prior unit-incremented 0xec or 0xed from an intuitive hex standpoint rather than possibly be a mistake as it was hard to understand all of the reported release dates / website updates, or understand the best sources of information etc.  when it was hard to know if some were abandoned.  It was an added burden.  That aided and abetted my confusion which caused my error of understanding that via Google brought me here.  But I'm not bitter or anything ... please don't misunderstand my intent ... I just want to be helpful!

Again - I'm saying all this because I really do, sincerely, appreciate the benefits of your hard work - it's good technology - it does what I want -  and I just want "more".  And I want better.  I want you to do really well (selfishly) so you can give me stuff that I want!  Carry on. :)