modmypi/PiModules #49: Safe shutdown triggered too soon/ too often

CarstenSchelp, 2019-09-18:
Hello,
I am searching for a way to configure how the pico UPS decides whether it should trigger a safe shutdown. Mine is a bit too jumpy, right now - shutting down when nothing is wrong, really (I have logs of the supply voltage). Is there a way to make this more inert?

PiModules, 2019-09-18:
Hi,

Please describe in more details your case for further support

If it is possible make a short video (10 seconds) of the system LEDs 
blinking, that will help me a lot

Kind Regards

CarstenSchelp, 2019-09-19:
> 
> Hi, Please describe in more details your case for further support If it is possible make a short video (10 seconds) of the system LEDs blinking, that will help me a lot Kind Regards
> -- Ioannis [www.pimodules.com](http://www.pimodules.com) -- Pi Modules Legal Disclaimer The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

Hi,

Thank you for picking this up. Do you mean putting on video just any random 10 seconds while the RPi is running? Because it only shuts down very occasionally it would be hard to catch the moment when this actually happens.

mminehanNZ, 2019-09-25:
I think he just wants a video of the LEDs when it's running normally.  The pattern of blinking of the BAT, CHG, UPS, INF will help him.
What is in the SysInfo register?  Look for 'SysInfo' in the manual if you have not seen it yet.  It does not clear on reboot, so you need to manually clear the register.  There's a script version in my fork that reads the SysInfo register.
I have two of these modules (HV3.0B+ PoE).  They are pretty powerful in terms of features, but difficult to set up and a little unreliable.  Certainly not a basic or beginner product.  The manuals need several errors corrected for one thing.
But keep trying.  Eventually you will sort it out.

mminehanNZ, 2019-10-21:
I have given up of PIco.  Too much trouble and not very helpful support.  See issue #48 ,