modmypi/PiModules #50: forum registration not working (reCAPCHA error)

XANi, 2019-11-07:
[image]

Also should probably use HTTPS...

XANi, 2019-11-14:
Yes, I've tried board's contact form and no response too

PiModules, 2019-11-14:
> forum

Hi,

In order to have better and more resistant forum to spams, we have had a 
lot, a lot of spams, we hired one person to move our forum, to dedicated 
server. He made it, but made it wrong, with know problems. Now we are in 
process to move our forum again to our old hosting address 
forum.pimodules.com

that will be available next days. I'm sorry for any inconvenience. We 
ware worry all that time than posts can be loosed.

BR Io

PiModules, 2019-11-14:
> That's still a thing? I had that problem at the beginning of the year too!

Hi,

In order to have better and more resistant forum to spams, we have had a 
lot, a lot of spams, we hired one person to move our forum, to dedicated 
server. He made it, but made it wrong, with know problems. Now we are in 
process to move our forum again to our old hosting address 
forum.pimodules.com

that will be available next days. I'm sorry for any inconvenience. We 
ware worry all that time than posts can be loosed.

BR Io