modmypi/PiModules #57: Problems installing PIco HV1.1 on Pi B+ with new install of Buster

RDK-45, 2020-03-27:
This Pi use to run the PIco HV1.1 with Wheezy as the OS,  it was retired several years ago.  I have just installed Buster (Linux 10) and would like to also activate the PIco HV1.1.

I have tried to follow my notes from when this unit was built with Wheezy, those in the "Simple_Setting_Guide_for_the_UPS_PIco.pdf" and those on the GITHUB site (which I think were written for the HV3 unit?).  In all cases it does not seem to be functioning.  For example the pico_status.py throws a Python error.  The output from the "sudo i2cdetect -y 1 " looks correct.

My original notes use the rc.local to initiate the picofssd.py program while the newer procedure sets it up as a service.

I have made a backup of my sd card which was taken before I tried any of the install procedures above.  Can you help?....RDK

PiModules, 2020-03-27:
Hi,

Hmm,

I must setup the workbench for the old PIco HV1.1 to check it on my 
side, however it is difficult to be done next 2 weeks as we are 
overloaded with new firmware testing for the PIco HV3.0A/B/B+ and new 
PIco HV4.0 release. If you can wait little bit, I will be able to work 
on this just after new PIco HV4.0 release

Kind Regards

Ioannis

RDK-45, 2020-03-27:
As I'm also waiting for the release of the HV4 unit, I'll be patient. Maybe someone else will have an idea?...RDK

RDK-45, 2020-03-27:
This was closed by mistake!!!

baahver, 2020-06-22:
> This Pi use to run the PIco HV1.1 with Wheezy as the OS, it was retired several years ago. I have just installed Buster (Linux 10) and would like to also activate the PIco HV1.1.
>
> I have tried to [...]

Are you sure this is not a python version error? I've had this before and noticed the change from python 2 to 3. running the script as:
`python2 pico_status.py`
was working for me.