modmypi/PiModules #58: ups pico hv3.0 b+ /raspberry Pi4

sevilla1092, 2020-04-16:
Is it possible to install global-en and run pin on raspberry pi 4? software is working ok, but I can't get it to reboot after a shutdown, only if I unplug the power cord and reconnect.

PiModules, 2020-04-16:
Hi,

The UPS PIco have been NOT designed for the Raspberry Pi 4 (s the Pi4 
was not available), however should be working on your Pi4 - here is 
working. The UPS Pico HV4.0 is designed especially for the Raspberry Pi4.

You need to use a small cables that connects the places of the  Gold 
Pin, and Power Enable (i.e. PE and P4) with places of the Raspberry Pi 4 
Enable and RUN. This should make your system working properly It s not 
necessary to use the PE (so one cable then is needed).

Also please decrease the speed of the I2C, to much lower rate i.e. 25K 
or even 10K, as the clock stretching is not working properly on the 
Raspberry Pi, so you need to add little bit more time for PIco I2C 
instead of stretching. We are working on a totally new I2C handler for 
the UPS PIco (for the HV3.0 - as the HV4.0 has different processor)

*/# Added for PIco
enable_uart=1
dtoverlay=i2c-rtc,ds1307
dtparam=i2c_arm=on
dtparam=i2c1_baudrate=25000/*

Hope it helps, please keep me informed.

Kind Regards

Ioannis

sevilla1092, 2020-04-16:
Thank you very much, I already made the modification and it is working excellent, I await news about HV4.0.

closed 2020-04-16 by sevilla1092

pozsothereal, 2021-11-14:
Hi.
Pretty old topic, btw i would interested about using UPS Pico HV3.0A on Rpi B+ with new Debian 11 based Bullseye rapbian, 32bit. Any thoughts? Should i update to bullseye to check?
Thx!
p.

PiModules, 2021-11-14:
Hi,

Now we have well working our own forum (without spams anymore), can you 
PLEASE move this question to this and I will answer in details

https://pimodules.com/community

We are trying to consolidate all info again on this forum, and much 
easier to handle than all others.

Thank you and Kind Regards

Ioannis

pozsothereal, 2021-11-14:
OKAY!

p.