modmypi/PiModules #59: UPS Pico HV3.0B+ , battery not charging, switching off power make the RPI dying.

EricPineau-56, 2021-01-21:


PiModules, 2021-01-21:
Hi,

Please provide more description on this, how and when it happens, what 
changes you done before, how old is batter etc. If you provide more info 
I will be able to support you.

Kind Regards

Ioannis/PiModules

EricPineau-56, 2021-01-21:
[image]
[image]

EricPineau-56, 2021-01-21:
[image]
it happens from the first boot after installation, each time I unplug the powersource, either on the micro usb or  on the ext power.
I then check the registers and found that the battery type was 0x43, so I set it to 0x53.
I also changed the battery timeout to 9min just to check.
But no change.
The OS was dowload from the OpenPlotter project, and I've followed the installation procedure in your documentation (v3.0 from 16/06/2018). I've installed the deamons and the RTC clock.

PiModules, 2021-01-21:
> IMG_20210121_205507

Hi,

Please at the first step use our OS image where everything is installed 
properly, and then come back

It is rather not possible that UPS PIco is faulty because we are testing 
each device before shipping

Kind Regards

Ioannis/PiModules

-- 
Ioannis Mourtsiadis
Electronic Engineer M.Sc

Pi Modules
Nikaias 49, Nea Smyrni
17122 Athens, Greece

Tel:    +30 211 0130067
Fax:    +30 211 8500132
Mobile: +30 697 7089430
SKYPE:  i_mourtsiadis

www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

EricPineau-56, 2021-01-25:
Hi, not knowing where your own distribution is located, I've installed the latest NOOB with the raspbian OS, then I've followed your documentation to install the ups.  After that, I've followed the procedure to Reset the ups pico to factory default (UR+C), then the  pico_status script told me ERROR on the line Battery version. I then set it to 0x53 (LIPO), the script now succeeds, but the battery voltage still at 0.3v.
So I'm at the same point than a few days before. 
What do you suggest ?
Regards
Eric
[image]
[image]

EricPineau-56, 2021-01-26:
A little precision: I've measured the voltage on the battery this morning with my voltmeter, and I measured 4.14V. So the battery is good.

PiModules, 2021-01-26:
Hi,

OK. Please follow my last recommendation about image, just to be sure 
that system is working properly and then together with you I will setup 
your own image as you need.

Kind Regards

Ioannis

EricPineau-56, 2021-01-26:
Hi, OK but could you give me the link to download, I couldn't find it? 
Thanks 

⁣Envoyé par BlueMail ​

PiModules, 2021-01-26:
Hi,

Please use this image from here,

https://drive.google.com/file/d/1wrRGv9n1vyp9NuTU9dg0GqzBKN1cyEz9/view

*User:* pi

*Password:* 1

This is 16GB NOOBS Version 3.4 with activated serial port for user, 
activatd i2c, activated SSH, activated 1-wire, and installed MC, 
minicom, set the Hardware RTC (on PIco) using old version of interaction 
(via GPIO – as before) – release, ready to use

Please try this and if OK, then we  will do together set up of your own

Warmest Regards

Ioannis

EricPineau-56, 2021-01-26:
Hi Loannis
I did all that this evening, but the situation is still the same. Tried to restore to factory default again, unsuccessfully. 
Should I have to upgrade the firmware, or anything else? 
Regards 
Eric 

⁣Envoyé par BlueMail ​

PiModules, 2021-01-26:
Hi,

Step by Step.

Lest specify in details you problem.

Please send me:

1. Detailed specification of your system (how you are powering it, what 
battery have you connected, what additional hardware is used etc)

2. Some short video with system LEDs

3. As much you think useful information

Then I will be able to help you, you must be doing something wrong, so 
we must fixed it based on received information.

BR Io

EricPineau-56, 2021-01-27:
https://user-images.githubusercontent.com/77807012/106060841-dfebee80-60f4-11eb-80c6-72b6f0ee53e3.mp4

here is the video, couldn't do better, cause of my very poor bandwidth.
The system is powered by the micro USB connector, but I also tried powering it with an external 12v source connected to the  ups directly, without change.
The battery i sthe one you've send me a few weeks ago, a LIPO 1000mAh.
As I told you in a previous message, I've tested it with my multimeter, the voltage was 4.14V.
The "Magic" switch  is now deconnected (to be sure there's no effect).
The  OS is the one downloaded using the link you gave me yesterday.
The video starts on a cold boot.
As you can see, when the system has finished the startup, the UPS Led blinks about 2 times/s, the BAT Led is lighting continuously.
The CHG Led is lighting at the end of the video, but after a minute or 2, this Led goes OFF.
The INF Led (name CHG on your documentation ?) is flashing (what does it mean ?)
The pico status report is the same I transmited to you yesterday or the day before.
No other hardware connected to the RPI nor to the ups.
MAybe something could help : When I powered it for the fist time 2 weeks ago, the battery was connected, the UPS pico was pluged on the RPI, and the system installed was OpenPlotter V2. Each one mn after the boot, the system was rebooting. So I turned it off, unplugged the ups pico, and restarted the system to apply the installation procedure described in your documentation. Once done, I truned the RPI off again, plugged the pico in, and restarted the system. The "after 1mn auto reboot" didn't appeared again, but the ups pico didn't work.

EricPineau-56, 2021-01-27:
I have to say that 3 years ago I already installed a UPS Pico on a RPI3, and it worked perfectly since the beginning, and still does. I'm surprised by these difficulties.

PiModules, 2021-01-27:
If something is wrong with PCB, I send you of course new one(s), and 
probably UPS PIco HV4.0B as we are waiting for the new production batch 
very soon. So, Please do not stress, we will do it working on any way. 
But lets try to make working your system first. Now is too late (and we 
are all closed in our home prisons), so I will check your messages 
tomorrow during the day and answer with probably more questions or solution.

BR Ioannis

EricPineau-56, 2021-01-27:
One more thing : when I try to reset the pico to fadctory default, it works , buut the battery type is then 0x40, that's weird because it doesn't match any battery type code described in your documentation ? And effectively, the pico status script output an error.
I then set it to 0x53 with i2cset command<and the pico status script then works ok.

EricPineau-56, 2021-01-27:
No problem, I'm confident we'll find a solution, and don't feel obliged to answer so late, if I write at that time it's because it's my rest time. I will understand if you answer during office hours, I know too much how support is difficult and sometimes painful. :-) 
Good evening 
Eric 

⁣Envoyé par BlueMail ​

EricPineau-56, 2021-02-15:
Hi Loannis, it's now 3 weeks since our last message, I'm ok to make eventually some new tests, but please tell me what. I'd appreciate to find a solution to that issue, and to be able to work use my rpi.
Thanks
Eric

PiModules, 2021-02-15:
> find a solution to that issue, and to be able to use my rpi.

Ohhh
Sorry
I apologize missed you as it was not in standard support email 
I will answer from my pc tomorrow morning
Sorry again for my fault 
BR io

⁣Get BlueMail for Android ​

PiModules, 2021-02-16:
Hi,

Apologize again for my late replying,

Can you please use a new battery 450 mAh that is included in the UPS 
PIco and confirm the behaviors

I'm trying t fix if it is battery matter or PCB or anything else

I m here fir you

BR Io

EricPineau-56, 2021-02-16:
Hi, I only have the battery you send me a few months ago now with my order. Its ref is LP603048, 1000mAh 3.7V. I have no other battery.
See the photo from 21 days ago above, and the comment I wrote about the voltage which appeared good when I measured it with my voltmeter.

²

PiModules, 2021-02-16:
Hi,

OK. So battery is fine. please confirm that UPS LED is blinking as also 
how fast is blinking

Kindly awaiting for your answer

BR Io

EricPineau-56, 2021-02-16:
Hi, an extract of my message from 20 days :
After a cold boot, when the system has finished the startup, the UPS Led blinks about 2 times/s, the BAT Led is lighting continuously.
The CHG Led is lighting at the end of the video, but after a minute or 2, this Led goes OFF.
The INF Led (name CHG on your documentation ?) is flashing (what does it mean ?)

No other hardware connected to the RPI nor to the ups.
MAybe something could help : When I powered it for the fist time 2 weeks ago, the battery was connected, the UPS pico was pluged on the RPI, and the system installed was OpenPlotter V2. Each one mn after the boot, the system was rebooting. So I turned it off, unplugged the ups pico, and restarted the system to apply the installation procedure described in your documentation. Once done, I turned the RPI off again, plugged the pico in, and restarted the system. The "after 1mn auto reboot" didn't appeared again, but the ups pico didn't work.

PiModules, 2021-02-16:
Hi,

As far the USP LED is not blinking slow or fast the daemons are not 
working properly.

Is it possible run the system with  SD image provided by me ?

Just to fix the reason,

Then we will go to your setup.

Please download it from here burn the SD and report the status, it is 16GB

https://drive.google.com/file/d/1wrRGv9n1vyp9NuTU9dg0GqzBKN1cyEz9/view

*User:*pi

*Password:*1

BR Io

EricPineau-56, 2021-02-16:
The UPS Led blinks 2 times /s, this is not good ?
Anyway, I've downloaded your image and loaded it on the SD card, 20 days ago, without any change or improvement.  Just a remark: from my memory, I wasn't asked for a any login nor password.

PiModules, 2021-02-16:
no good, sorry, but most possible is wrong setup

BR Io

EricPineau-56, 2021-02-16:
so you're checking if the image is good ?

EricPineau-56, 2021-02-17:
what do you want me to do ?

PiModules, 2021-02-17:
Hi,

Just burn the image I sent you to 16GB card (or 32) and run the UPS PIco 
+ Raspberry

and report me the results about UPS LED (how and if it is blinking - no 
blink, fast or slow)

We are doing this in order to be sure that you made a proper 
installation of the daemons.

If this image will be not working properly (i do not believe it, but 
if), the the problem is in another point

However I mostly sure the problem is in the installation you made

Please check it and inform me

BR Io

EricPineau-56, 2021-02-17:
ok, I'll do it this evening

PiModules, 2021-02-17:
Keep your time,

BR Io

EricPineau-56, 2021-02-17:
done, nothing new, still the same : UPS led blinking 2 times/s, but when I unplug the mico USB power supply from the rpi, it dies immediately, no safe shutdown.
Joined a photo from the pico_status.
[image]

PiModules, 2021-02-17:
> IMG_20210217_222532

how many units you have ?

EricPineau-56, 2021-02-17:
Only one

⁣Envoyé par BlueMail ​

PiModules, 2021-02-17:
From production we received I have some spare units, so can send you 
end of next week, or you can wait and receive the UPS PIco HV4.0B that 
should be here in about 2 max 3 weeks

Probably something is not properly soldered ( the J2 connector), but is 
is useless to ask you to solder their pins

Each PCB is tested here with all critical functionalities (like this you 
motioned), however for testing we are using POGO pins that are attaching 
the PCB on the bottom, so if the J2 connector is not soldered properly 
it is not possible to recognize. Until now only few times it happens 
(2-3 times) in many thousand of PICO we sold.

As we ware expecting to have much earlier the new version for UPS PIco 
HV4.0B, but due to Covid-19 workers infection in the factory in Poland, 
we lost our time schedule. So, we are totally out of units. We received 
a lot of devices but all are sold out for industrial customers. However 
I have some spare from this production so can cover you with one device

I will be able to send it at the end of the testing process end of next 
week (more or less), or you can wait so we will have here many, many 
units if the UPS PIco HV4.0B and then can cover you

Decision is yours

Please advise with your details to *support@pimodules.com*

Of course I do _*not need*_ the old device, I can help you to repair it, 
but I send you new one with our costs.

BR Io

Due to that we

EricPineau-56, 2021-03-29:
hi Loannis, it's now 1,5 month later, and still no news from you, getting a bit anxious to get my working UPS pico, since this affair lasts for a few monthes now... Could you send me the new one you promised me ? (notice I don't mind to get the v4)
thanks
Eric

PiModules, 2021-03-29:
Hi,

Of course I will send you, with extra Super Capacitor 100F and FAN as 
promised. But due to Covid we have had a strong delay in manufacturing, 
as worker in the factory was infected many, many times. However PCBs 
already arrived and we are under very strong testing process as it is a 
new product + a very complicated product, there are 250 parts on this 
small PCB. So we are doing the testing of all PCBs, before start 
shipping. If you cannot wait, then I can cover you with the old version 
as we receive one production LOT of the old HV3.0B+, and have some are 
spare. We are here also under hard lock down, so testing is distributed 
between various places, so takes longer. I expect that I will be able to 
cover you in about 2-3 weeks from now.

It is up to you, please advise what you prefer, the old one or the new 
one with free add-ons.

Stay Healthy !!

My Warmest Regards

Ioannis

EricPineau-56, 2021-03-29:
Hi Loannis,
thanks for your answer, an old version would be ok for me, still, it's what I baught ;-) 
Eric

PiModules, 2021-03-29:
please reminded me what mode do you have (stack or Advanced)

as also send me your address to support@,

BR Io

EricPineau-56, 2021-03-29:
[image]

PiModules, 2021-03-29:
I got it, but please delete this email, as there is not reason to 
publish your personal data over the public forum

BR Io

EricPineau-56, 2021-05-07:
Hi loannis, still more than 1 month later, non news from you, neither any new UPS Pico received. Seems my patience's running out, I start feeling treated as an idiot. So now I ask you to send me a working UPS Pico next week. I've ordered my product in november, and you've been making me waiting since january, postponing the resolution of the problem 1 month after the other. As a professional engineer working in industry, I have to say it's not a way to be with customers. Waiting for a quick answer...

PiModules, 2021-05-07:
Hi,

We are working hard with received production of new UPS PIco HV4.0B, 
just due to volume of units and very extensive test we are slow. Much 
slower than expected. Also due very long lock down, we are working in a 
separate places, so takes much more time than expected

As I promised and of course keep my promise, to send you replacement 
with the new PIco + extra parts for free, not asking for return. Faulty 
units happen, rarely on our products but happen, on all companies. We 
accepted your claim and of course going to replace the unit with more 
expensive more advanced + extra parts.

But please understand that until mid next week we ware working under not 
normal conditions (remotely) in separate places, that causes delays. But 
this was the only solution to avoid Covid infection.

Please note that in Greece we have every day about 100 people dead due 
to corona (Greece is 10 millions) so situation is really not normal. 
Most of them are in Athens. So, we must work very carefully and avoid 
common working on the same places. Within this month we all will be 
vaccinated so this black story will be ended at all. We will work again 
in the same place.

Everything will be closed within this months, we are near to close 
everything and starting packaging and shipping also your replacement. I 
assume that in France is similar

Thank you for understanding and patience

Stay healthy !!

Kind Regards

Io