modmypi/PiModules #60: Stuck unable to upload firmware - getting "\xfc" from /dev/ttyAMA0

garethhowell, 2021-08-09:
I tried to upload newer firmware on to my HV 3.0A and I seem to have bricked the Hat.

I initially tried to upload `0xE3_UPS_PIco_HV3.0_main.hex` but got an error part-way through the error. Unfortunately, I failed to record the error.

Since that time I have been unable to communicate with the bootloader. I get
```
$ sudo python 9600_picofuHV3.0.py -f 0xE3_UPS_PIco_HV3.0_main.hex
Validating firmware: OK
Checking communication with bootloader: KO
ERROR: Invalid response from PIco: 252  Please retry the FW upload process.
KO
ERROR: Something wrong happened during verification of communication channel with PIco bootloader via port: /dev/ttyAMA0 Please verify that the serial port is availble and not used by some other application.
```
I have repeated this several times, but get the same result. If I use minicom I can see that the bootloader is returning `\xfc`

Any ideas?
*(I wanted to ask this in the PiModules forum but it seems to be inactive - can't register and get no response from support)*

PiModules, 2021-08-09:
Hi,

You can use the latest upgraded and updated Raspberry Pi image, just in 
a case

https://drive.google.com/file/d/1QE4BujEKHZEWSTiAWR5Ebf2sP6nTEEP-/view?usp=sharing

For image installation you can use this one

https://sourceforge.net/projects/win32diskimager/

*user:*pi

*pasword:*1

Also you can use the latest firmware, we not released it yet officially

use for upload one of below commands on one line (for GIPO or without GPIO):

sudo sudo i2cset -y 1 0x6b 0x00 0xff && sleep 2 && python 
serial0_9600_picofuHV3.0.py -v -f 
UPS_PIco_HV3.0_main_29_120721_GPIO_FREE.hex

sudo sudo i2cset -y 1 0x6b 0x00 0xff && sleep 2 && python 
serial0_9600_picofuHV3.0.py -v -f 
UPS_PIco_HV3.0_main_29_120721_GPIO_USED.hex

Download both on one zip from here:

https://pimodules.com/download/ups_pico_hv3-0_main_29_120721

Changes made for public:

1. Delayed start and shutdown in seconds

/sudo i2cget -y 1 0x6B 0x03/ *to check (default is 0)*

/sudo i2cset -y 1 0x6B 0x03 10/ *to set to 10 seconds (up to 255)*

2. Charger ON, OFF and ON during system on LPR

/sudo i2cget -y 1 0x6B 0x1d/ *to check (default is 0xFF) for ON
*

/sudo i2cset -y 1 0x6B 0x1d 0x00/ *to set OFF*

/sudo i2cset -y 1 0x6B 0x1d 0xCC/ *to set ON also when system is in LPR*

3. How long system will be running without response from Raspberry Pi 
(Very usefully when doing update or upgrade on OS) in minutes

/sudo i2cget -y 1 0x6B 0x1f/**to check (default is 0x02) for 2 minutes**

/sudo i2cset -y 1 0x6B 0x1f/**/0x05 /**to set for 5 minutes without 
doing shutdown (and going to LPR)**

When you sent email to support, I can not see it, Please send it again, 
need to check if arrived

Best Regards

Ioannis/PiModules

garethhowell, 2021-08-10:
Thanks for the speedy response.

1. The Pi was built with the latest NOOBS image.

2. The software method for entering bootloader does not work because there is no valid firmware installed. I used the hardware method.

3. The problem is failed communications with the bootloader. Here is a dummy reproduction of the relevant portion of 9600...

```
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import serial, io
>>> s = serial.Serial(port="/dev/ttyAMA0", baudrate=38400) # Note: 38400 Baud
>>> s.write(":020000040000FA\r")
16
>>> s.in_waiting
8
>>> s.read(8)
' \xfc \xfc \xfc \xfc'
```
The uploader expects '\x0a'

NB. If I use 9600 Baud for the port speed, there is no response from the bootloader.

PiModules, 2021-08-10:
Hi,

Please send me email, you set to support I must fix the problem as we 
not answered to you

OK. The latest image I sent you also by a link (NOOBS also)

Please send me list what serial ports are available to check by Linux 
command

Do you have any hardware or application using the serial port ?

BR Ioannis

garethhowell, 2021-08-10:
Thanks, Ioannis
Here’s a screenshot of the serial ports.

[A picture containing background pattern  Description automatically generated]

There is nothing connected to the Pi, and no processes using the ttyAMA0 as far as I can tell.
[Text  Description automatically generated]

The email to support was sent through the forum website, so I don’t have a copy.

Gareth

Reply to: modmypi/PiModules ***@***.***>
Date: Tuesday, 10 August 2021 at 14:17
To: modmypi/PiModules ***@***.***>
Cc: Gareth Howell ***@***.***>, Author ***@***.***>

PiModules, 2021-09-11:
> Is there any progress on this? At the monent, the UPS is completely
> dead and I can get no response from it beyond what I have indicated
> above. Is it dead? Is there any way to reinstate the boot loader?

Hi,

The bootloader is in protected part of memory so can not be destroyed in 
any "normal" way

I giving below some links for the new firmware, script uploading the 
firmware, as also attachments

If your PIco is not responding to the software request, just use the 
hardware, but keep the delay if the 1-2 seconds

like this, bootloader must be invoked by hands (with two buttons)

_*if hardware invoked*_

sleep 2 && python serial0_9600_picofuHV3.0.py -v -f 
UPS_PIco_HV3.0_main_29_120721_GPIO_FREE.hex

or

sleep 2 && python serial0_9600_picofuHV3.0.py -v -f 
UPS_PIco_HV3.0_main_29_120721_GPIO_USED.hex

_*if software invoked*_

sudo sudo i2cset -y 1 0x6b 0x00 0xff && sleep 2 && python 
serial0_9600_picofuHV3.0.py -v -f 
UPS_PIco_HV3.0_main_29_120721_GPIO_FREE.hex

sudo sudo i2cset -y 1 0x6b 0x00 0xff && sleep 2 && python 
serial0_9600_picofuHV3.0.py -v -f 
UPS_PIco_HV3.0_main_29_120721_GPIO_USED.hex

Also try to use my image to be sure about everything

Image 32GB

user : pi

pass: 1

https://drive.google.com/file/d/1QE4BujEKHZEWSTiAWR5Ebf2sP6nTEEP-/view?usp=sharing

For image recovery you can use this one

https://sourceforge.net/projects/win32diskimager/

_*Changes made for public:*_

1. Delayed start and shutdown in seconds

/sudo i2cget -y 1 0x6B 0x03/ *to check (default is 0)*

/sudo i2cset -y 1 0x6B 0x03 10/ *to set to 10 seconds (up to 255)*

2. Charger ON, OFF and ON during system on LPR

/sudo i2cget -y 1 0x6B 0x1d/ *to check (default is 0xFF) for ON
*

/sudo i2cset -y 1 0x6B 0x1d 0x00/ *to set OFF*

/sudo i2cset -y 1 0x6B 0x1d 0xCC/ *to set ON also when system is in LPR*

3. How long system will be running without response from Raspberry Pi 
(Very usefully when doing update or upgrade on OS) in minutes

/sudo i2cget -y 1 0x6B 0x1f/**to check (default is 0x02) for 2 minutes**

/sudo i2cset -y 1 0x6B 0x1f/**/0x05 /**to set for 5 minutes without 
[... 1376 lines cut ...]
:10643000F28BBF0000F0A60003003700040020002C
:10644000935F800003005200800A78004F037800B9
:10645000CF02780000000600851F7800845F80006E
:10646000E502420005027800A300200011000900A7
:106470000382D800F0ABB700845F80006A20B800C8
:10648000F0ABB700F00BF8000000F900CF02780085
:1064900000000600FAABEF00F40BF800F60BF90071
:1064A000F82BEF00FA0BE2002E003200FA2BED0081
:1064B000B45F8000F37022000418E1000200310094
:1064C00000A0EF0029003700B45F8000833E200069
:1064D0000418E10008003E00B45F8000843EB10073
:1064E000B45F880004002100C35F80000480410085
:1064F000F8ABB7001A003700B45F800043062000F5
:106500000418E10008003E00B45F80004406B100BA
:10651000B45F880004102000C35F80000480410045
:10652000F8ABB7000E003700B45F80006A20E100CE
:1065300007003900B45F80006A005200F6ABB70074
:10654000C45F800070004200F8ABB7000400370061
:10655000F88BBF00F60BB400F8ABB70001003700B2
:10656000D0FF3700F80BF8000000F900000006002B
:10657000851F78002A0BF800F00BF9002C3202007E
:10658000000000002AABB7002C2BEF002A0BF8000C
:10659000F00BF9002C3202000000000080027800AD
:1065A000A55F88004A3202000000000000A0B7008A
:1065B

PiModules, 2021-09-12:
_*Potwierdzam, *_

sa procesory, ale o 1.5 euro drozsze wiec dodatkowe sztuki beda o 1.5 
euro drozsze

Jezli potwierdzsz ze tak, to je kupuje teraz (bo moga zniknac), a MUSISZ 
zaplacic jutro rano

Prosze o natychmiastowe potwierdzenie

dziekuje

Pozdro

Io

-- 
Ioannis Mourtsiadis
Electronic Engineer M.Sc
--
Pi Modules
Raspberry Pi Approved Design Partner
Nikaias 49, Nea Smyrni
17122 Athens, Greece

Tel:    +30 211 0130067
Fax:    +30 211 8500132
Mobile: +30 697 7089430
SKYPE:  i_mourtsiadis

www.pimodules.com
--

Pi Modules Legal Disclaimer
The content of this email is confidential and solely intended for the indicated addressee. Access to this email by anyone else is unauthorized. If you are not the intended recipient be informed that disclosure, reproduction in any form, in part or whole, or circulation of the content of this email is prohibited and may be unlawful. In this case, please inform immediately the sender of the email.

PiModules, 2021-09-12:
wrong post has been sent - sorry

garethhowell, 2021-09-13:
Thanks, Ioannis
Using your image seems to have fixed it. I now have 0x29 running and I can now control the fan properly - which I couldn't on the old firmware.

closed 2021-09-13 by garethhowell

PiModules, 2021-09-13:
Hi,

Good !

I strongly believe that today we will lunch the new forum, so support 
will be much easier for me

Lest hope, it is already running on copy website, and just waiting to 
our subcontractor to move it to official location

BR Ioannis

LouDnl, 2023-06-11:
Can you or anyone make this [pimodules.com/download/ups_pico_hv3-0_main_29_120721](https://pimodules.com/download/ups_pico_hv3-0_main_29_120721) download available??