modmypi/PiModules #62: UPS stops working (UPS LED OFF) after some minutes, even if on Power.

garethhowell, 2021-09-13:
This is really a follow-on to [https://github.com/modmypi/PiModules/issues/60](https://github.com/modmypi/PiModules/issues/60).

System is Pi 3B running Ioannis' image downloaded from Google Drive.
Firmware is `UPS_PIco_HV3.0_main_29_120721_GPIO_USED.hex` as supplied on the above image.
No settings have been changed.

System starts OK and seems to be running: UPS LED is ON, all others are OFF
```
$ python pico_status.py 
 
***********************************
      UPS PIco HV3.0A Status       
***********************************
 
  UPS PIco Firmware.......: 29
  UPS PIco Bootloader.....: 53
  UPS PIco PCB Version....: 41
  UPS PIco BAT Version....: LiPO (ASCII: S)
  UPS PIco BAT Runtime....: 2 MIN
  UPS PIco STA Time Left..: 127 SEC
  UPS PIco r232 State.....: OFF
 
  Powering Mode...........: RPi POWERED
  BAT Percentage..........: 93.3333333333 %
  BAT Voltage.............: 4.1 V
  RPi Voltage.............: 5.4 V
  NTC1 Temperature........: 33 C
  TO-92 Temperature.......: 00 C
  Extended Voltage........: 14.3 V
  A/D2 Voltage............: 0.0 V
 
  PIco FAN Mode...........: AUTOMATIC
  PIco FAN State..........: OFF / AUTO
  PIco FAN Speed..........: 50 %
  Pico FAN Temp Threshhold: 50 C
 
***********************************
         Powered by PiCo           
***********************************
```
```
$ systemctl status picofssd
● picofssd.service - LSB: PiModules(R) UPS PIco file-safe shutdown daemon
   Loaded: loaded (/etc/init.d/picofssd; generated)
   Active: active (running) since Mon 2021-09-13 17:29:09 BST; 9min ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 2 (limit: 871)
   CGroup: /system.slice/picofssd.service
           └─776 /usr/bin/python /usr/local/bin/picofssd --log-level debug --pid-file /var/run/picofssd.pid --xml-config /etc/pimodule

Sep 13 17:29:07 raspberrypi systemd[1]: Starting LSB: PiModules(R) UPS PIco file-safe shutdown daemon...
Sep 13 17:29:09 raspberrypi systemd[1]: Started LSB: PiModules(R) UPS PIco file-safe shutdown daemon.
Sep 13 17:29:24 raspberrypi picofssd[776]: <INFO>: Started
Sep 13 17:29:24 raspberrypi picofssd[776]: <DEBUG>: False
Sep 13 17:29:24 raspberrypi picofssd[776]: <DEBUG>: None
Sep 13 17:29:24 raspberrypi picofssd[776]: <DEBUG>: None
Sep 13 17:29:24 raspberrypi picofssd[776]: <DEBUG>: None
Sep 13 17:29:24 raspberrypi picofssd[776]: <DEBUG>: None
```

However, two things:
If I remove power, I then get:
UPS LED stays ON, and BAT LED stays OFF. And
```
$ python pico_status.py 
 
***********************************
      UPS PIco HV3.0A Status       
***********************************
 
  UPS PIco Firmware.......: 29
  UPS PIco Bootloader.....: 53
  UPS PIco PCB Version....: 41
  UPS PIco BAT Version....: LiPO (ASCII: S)
  UPS PIco BAT Runtime....: 2 MIN
  UPS PIco STA Time Left..: 127 SEC
  UPS PIco r232 State.....: OFF
 
  Powering Mode...........: ERROR
  BAT Percentage..........: 80.0 %
  BAT Voltage.............: 4.0 V
  RPi Voltage.............: 5.2 V
  NTC1 Temperature........: 32 C
  TO-92 Temperature.......: 00 C
  Extended Voltage........: 13.9 V
  A/D2 Voltage............: 0.0 V
 
  PIco FAN Mode...........: AUTOMATIC
  PIco FAN State..........: OFF / AUTO
  PIco FAN Speed..........: 50 %
  Pico FAN Temp Threshhold: 50 C
 
***********************************
         Powered by PiCo           
***********************************
```

(NB I enabled loglevel=debug)

and, after 5 minutes, the UPS LED goes OFF and stops operating.
```
$ python pico_status.py 
 
***********************************
      UPS PIco HV3.0A Status       
***********************************
 
  UPS PIco Firmware.......:
Traceback (most recent call last):
  File "pico_status.py", line 206, in <module>
    print " ","UPS PIco Firmware.......:",fw_version()
  File "pico_status.py", line 27, in fw_version
    data = i2c.read_byte_data(0x69, 0x26)
IOError: [Errno 121] Remote I/O error
```
Pressing `IR` causes UPS to restart for another 5 mins.

PiModules, 2021-09-13:
Please follow this case on our forum, from tomorrow,

I give you  a note when ready and uploaded.

I will ask you for video etc

BR Io

garethhowell, 2021-09-14:
Thanks, Ioannis
I have a video ready and waiting

Gareth

PiModules, 2021-09-14:
Hi,

I already asked company doing it for us to upload the parallel system to 
the public website, should be ready by today, so the we will be able to 
use everything as proper

Please wait, I will announce it here also

BR Io

PiModules, 2021-09-14:
Hi,

It is uploaded (new forum), will be under testing phase for some hours.

Can you please try to register to new forum, as a totally independent 
person. Sorry for using you as testing person.

Any comments please send to ***@***.***

https://pimodules.com/community

We will support only via our own forum.

Thank you and Kind Regards

Ioannis